RobotWebTools/rosbridge_suite #160: rosbridge deb appears to still be broken in indigo shadow-fixed

mitchellwills, 2015-01-04:
The latest version of rosbridge server (0.7.6) that is in shadow-fixed appears to be broken. I am unable to connect to the websocket server using roslibjs. The library attempts to connect to the server (sends request headers), but the server never responds (checked with wireshark). This was tested on 14.04 with firefox and chrome. The same setup works on 0.7.0 and also appears to work up to 0.7.4, but fails on 0.7.5 and 0.7.6. Also at one point in my testing I think it managed to break my global tornado installation (it caused another package I wrote that uses tornado to exhibit similar symptoms), but uninstalling fixed it.

rctoris, 2015-01-04:
Confirmed. @jihoonl while intentions for your PR were good in theory, I doesn't appear a good clean solution is working once we get to the debian packages. It would be pretty bad if this makes it into the upcoming sync so I'm thinking we might need to revert once again. Quick thoughts anyone else?

rctoris, 2015-01-04:
Also, @jihoonl should we notify Tully to hold the sync once again for this?

jihoonl, 2015-01-04:
arg.............it seems like my test was inappropriate. I have confirmed that rosbridge_server was running without any crash. I should have tested with examples in roslibjs to check whether it actually works...

jihoonl, 2015-01-04:
I have asked Tully to postpone upcoming sync. I will report back after testing.

rctoris, 2015-01-05:
Thanks @jihoonl !!

jihoonl, 2015-01-05:
It seems like there is no easy way to use tornado as sub module of `rosbridge_tools` unless we change the source code. We should revert back again... my bad.

jonbinney, 2015-01-05:
Is it not importing the sub module version of tornado?

jihoonl, 2015-01-06:
The problem is tornado does not accept handshake from client(e.g. roslibjs).
I have confirmed that some modules are loaded. But I couldn't figure out where it handles incomings and why it does not accept handshakes.

jihoonl, 2015-01-06:
I have kept the version 0.7.6 as separate branch so we can continue to test later.

https://github.com/RobotWebtools/rosbridge_suite/tree/tornado_magic_076

jihoonl, 2015-01-06:
0.7.7 is now in shadow-fixed. I confirmed that simple.html example works fine now.

closed 2015-01-06 by jihoonl