RobotWebTools/rosbridge_suite #161: Receiving messages from subscribed topic silently fails if any string contains a degree symbol

mitchellwills, 2015-01-04:
I am trying to subscribe to a diagnostics message that contains a degree symbol, but it silently dropped in rosbridge server. The problem appears to be with the json.dumps call here: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/protocol.py#L263 as it raises an exception "'utf8' codec can't decode byte 0xb0 in position 14: invalid start byte". I'm not super familiar with string encodings, but when I print out the message (as a python dictionary before the call the string is displayed like this: 'Temperature Critical (\xb0C)' where \xb0 is the hex escape sequence for the symbol. The string is originally created in c++ uisng the string literal "Temperature Critical (\xB0""C)" and is correctly displayed in rqt_robot_monitor. If I change the line mentioned above to the following then it works, but i'm not sure what the implications of this change are.
return json.dumps(msg, encoding='latin1')

mvollrath, 2015-01-12:
Seems like you could send the utf-8 code for the symbol?

According to http://www.utf8-chartable.de/ this is  `U+00B0` or `c2 b0`

You might try `\xC2\xB0` and see how that goes.

mvollrath, 2015-01-12:
Otherwise a good answer here http://stackoverflow.com/questions/17103925/how-well-is-unicode-supported-in-c11

mitchellwills, 2015-01-24:
That appears to have fixed the issue.

closed 2015-01-24 by mitchellwills