RobotWebTools/rosbridge_suite #162: find out how to use tornado as submodule of rosbridge_tools

jihoonl, 2015-01-05:
For better uses,

Related PR & issues 
* https://github.com/RobotWebTools/rosbridge_suite/pull/157
* https://github.com/RobotWebTools/rosbridge_suite/issues/154
* https://github.com/RobotWebTools/rosbridge_suite/pull/155

Branchs
* https://github.com/RobotWebTools/rosbridge_suite/tree/rosbridge_tools_update
* https://github.com/RobotWebTools/rosbridge_suite/tree/tornado_magic_076

jihoonl, 2015-01-05:
As reference,

rocon uses a fork or redis-py as submodule of rocon_python_redis package. The local changes are necessary to use as submodule.

**Example**
Original : https://github.com/andymccurdy/redis-py/blob/2.6.2/redis/__init__.py
Rocon fork : https://github.com/robotics-in-concert/rocon_tools/blob/indigo/rocon_python_redis/src/rocon_python_redis/__init__.py

jihoonl, 2016-05-06:
Should we keep the tornado in rosbridge_server in kinetic or separate it out?

asmodehn, 2016-05-23:
Please separate it out :)

I recently stumbled on this, while checking dependencies for https://github.com/asmodehn/celeros (and eventually https://github.com/asmodehn/rostful)
I had a tornado python package installed in ros indigo, and I had no idea where it came from for a long while.

The system (ubuntu) tornado package is also too old for me (I need >3.2), so I would need a tornado dependency as ros package. Following the ros distribution strategy, we should find a version that work for all projects that use it, and turn it into a ros package...

We should also get used to specify version requirements in our package.xml to make these explicit, like this : https://github.com/asmodehn/pyros/blob/config_refactor/package.xml

BTW : If you have a look at celeros and rostful you will see that I use submodules for dependencies : 
https://github.com/asmodehn/rostful/tree/indigo-devel/deps

I will, in the next few weeks, simplify these submodule dependencies and instead make ros packages for all the dependencies, using https://github.com/asmodehn/catkin_pure_python : it makes creating catkin packages from recent (setuptools) python packages much easier ( no need to modify setup.py, only CMakeLists.txt and package.xml are needed ).
It is still in development though, but feel free to try it, and contact me if anything causes you trouble.

asmodehn, 2016-08-12:
I had a look at tornado versions on ubuntu.

In my case I only need to make a ros package to get tornado 4.2.1 on trusty, utopic, and vivid.
That is for indigo and jade.
Later ubuntu & ROS distros should be fine using just a rosdep key referring to the system package.

@jihoonl Let me know which version of tornado you want me to release, and where we should put the release repo.

Cheers !

asmodehn, 2016-08-12:
Oh and the nice thing for third party releases : https://travis-ci.org/asmodehn/click-rosrelease/jobs/151744100

jihoonl, 2016-08-15:
@asmodehn  

I have create experimental-tornado branch where we can test with third party release tornado.

*Things to be discussed*
* How to apply [setup.py](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/setup.py) changes in new third party tornado deb.
* How to switch different tornado debs.
* Maintenance. How shall we maintain tornado as third party Debian package.

asmodehn, 2016-08-16:
- How to apply setup.py changes in new third party tornado deb.

    _- using catkin_pip there are NO setup.py changes required._

- How to switch different tornado debs.

    _- using package.xml, declaring a depend on "python-tornado" will get the system dependency via rosdep key._

    _- using package.xml, declaring a depend on "tornado" will get the ROS thirdparty package._

- Maintenance. How shall we maintain tornado as third party Debian package.

    _- we only need to maintain the release repository. Release new tornado versions when required, verify travis checks passes, apply some patches if needed, and thats it._

The only downside is that we need to manage all tornado dependencies as well.
We also rely on [catkin_pip](https://github.com/asmodehn/catkin_pip) to make thirdparty releases easy, but it does mean we need to (get familiar with it and) maintain it.

asmodehn, 2016-08-19:
I did a ROS pkg indigo release of **python-certifi**. First step. 

- https://github.com/asmodehn/certifi-rosrelease
- https://travis-ci.org/asmodehn/certifi-rosrelease/branches

Tornado [has some other dependencies](https://github.com/tornadoweb/tornado/blob/master/setup.py#L126) we (maybe) need to release for ROS as well : 

- **backports_abc**
- **singledispatch**
- **backports.ssl_match_hostname** (probably needed only for indigo. need to double check python versions)
- **ordereddict** (probably not needed on trusty since python > 2.7.0)

asmodehn, 2016-10-07:
I did all these a while ago, now it s time to do tornado ros package. But I didn't have time to work on it yet...

asmodehn, 2016-11-09:
Actually tornado 4.2.1 only needs: 
- **[certifi](https://travis-ci.org/asmodehn/certifi-rosrelease)**
- **[backports_ssl_match_hostname](https://travis-ci.org/asmodehn/backports.ssl_match_hostname-rosrelease)**

Already released on indigo : 
- http://repositories.ros.org/status_page/ros_indigo_default.html?q=certifi
- http://repositories.ros.org/status_page/ros_indigo_default.html?q=backports

I noticed **backports_ssl_match_hostname** is [duplicated in this repo](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosbridge_server/src/backports) as well.

Replacing it by a dependency and making sure things still work should probably be the first step...
[
tornado third party release](https://travis-ci.org/asmodehn/tornado-rosrelease) is coming soon...

asmodehn, 2016-11-10:
And done : http://repositories.ros.org/status_page/ros_indigo_default.html?q=tornado

Some of the travis tests are failing because travis platform doesn't support ipv6. The rest of them looks fine.

So @jihoonl, please try that package when you have some time, and let me know how it goes.
It s only available for indigo at the moment. 
When we are happy with it I ll release for jade (probably very little change) and kinetic.

Enjoy !

asmodehn, 2016-12-27:
@jihoonl any news about this ? tornado package has been released.
But it cannot be used while rosbridge also installs tornado...

asmodehn, 2017-01-31:
I recently registered a conflict in https://github.com/asmodehn/backports.ssl_match_hostname-rosrelease/ and in https://github.com/asmodehn/tornado-rosrelease.

Feel free to PR in there if that conflict declaration creates problems for you.

closed 2021-09-19 by github-actions[bot]