RobotWebTools/rosbridge_suite #165: Leaked subscriber connections (memory / thread leak)

rethink-rlinsalata, 2015-01-08:
_This is connected to the rospy issue ros/ros_comm#544 - please see more detailed info there._

There is an intermittent issue where a Subscriber's underlying TCP socket connection thread and objects (`rospy._SubscriberImpl` and `rospy.TCPROSTransport`) can be leaked and never cleaned-up, nor used again when new Subscribers for the same topics are created later on.  This means there is an occasionally occurring, but ever-increasing number of threads running TCP callback loops (end up being NOPs) and leaked memory objects, that increase the memory and usage of the `rosbridge_websocket` server process ***AND*** the Publisher node on the other side of the TCP connections to which the server is connecting, until the machine runs out of memory and kills one of them.  When this happens with our robot, the kernel oom_killer likes to choose our MotorController process to kill, thus, the issue makes our web-apps using rosbridge somewhat impractical... =/

These leaked connections seem to occur because of a race condition where a new connection (`TCPROSTransport`) is being added to a shared per-topic subscription implementation (`_SubscriberImpl`) that is already closed / being closed. In fact, you can add a debug check in the `_SubscriberImpl.add_connection()` function to see that it is already closed by the time the new connection is attached to it.

Unfortunately, the original, internal use case where this occurs (quite reliably, though still frustratingly intermittently), uses a piece of internal code that interfaces to rosbridge and touches a number of complicated components -- and my attempts to replicate a test case in an parred down, isolated manner just cannot reproduce the issue as reliably as when targeting a running robot and/or using the internal piece of code.  It does seem to happen most when there is a rapid barrage of new Subscribers and unsubscriptions (even, or perhaps specifically to only a limited number of the same topics), and also possibly when the ROS Publisher node that the topics come from is running at high-load.  My attempts to replicate the leak with just standard rospy Subscribers and Publishers (w/o going through rosbridge) have all been unsuccessful so far, but it is hard to produce an exact 1:1 equivalent test program.

The leak itself occurs in rospy, and I have made a ticket there - ros/ros_comm#544.  I believe I even have a "fix" that will catch the symptom of it breaking and prevent the leak-bit; however, _I do not know what is **triggering** the corrupt behavior._  I also do not know if whatever is triggering the improper connection addition/Subscribtion removal at a higher-level is breaking in its own way, or will break after the "fix."  Since I have only seen this in the scenario where rosbridge is triggering the leak, I wanted to post the issue in both places.

I also was not able to get a full understanding on all the pieces where the rosbridge library manages the subscriber objects and callbacks, and some of the code concerned me if the server was using the rospy API in an unexpected way, and thus exposing and triggering the leak.  For instance, could a race condition be being exposed in the explicit deletion of Subscribers, or in the parts that reach down and directly access the `_SubscriberImpl` object instead of using the get_impl() function of the Topic Manager (unlikely here, but for example: `MultiSubscriber.subscribe()` [subscribers.py~L117](https://github.com/RobotWebTools/rosbridge_suite/blob/bf6e9bcc9d6b12386785cb97af6231155deff46f/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L117) ).

Thanks for whatever insight / sanity-checks you have!

jihoonl, 2015-01-12:
I wonder if it is similar symptom we identified in https://github.com/RobotWebTools/rosbridge_suite/issues/138.

The symptom is that some topic msgs were lost when rosbridge unregister and reregister Publisher rapidly on the same topic name. Then subscribers couldn't receive topic messages which is from newly registered publisher for 2~3 minutes.

> [WARN] [WallTime: 1414462064.711620] Could not process inbound connection: [/chatchat_pub] is not a publisher of [/chatter]. Topics are [['/rosout', 'rosgraph_msgs/Log']]{'message_definition': 'string data\n\n', 'callerid': '/rostopic_32229_1414462060102', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/chatter', 'type': 'std_msgs/String'}

Also I believe it is introduced within a year because I couldn't see this behavior when I was using hydro.

mvollrath, 2015-09-16:
I've done some work on this and may have a solution.

Here's a test script that reproduces the leakage and warning messages:

    #!/usr/bin/env python
    
    from websocket import create_connection
    import socket
    
    NUM_CYCLES = 10
    WS_URL = "ws://localhost:9090"
    
    def test_cycle(count):
        print "running cycle {}".format(count + 1)
    
        ws_alpha = create_connection(WS_URL)
        ws_bravo = create_connection(WS_URL)
    
        ws_alpha.send(r'{"op":"advertise","topic":"/test_rosbridge_leak","type":"std_msgs/String"}')
    
        ws_bravo.send(r'{"op":"subscribe","topic":"/test_rosbridge_leak","type":"std_msgs/String"}')
    
        ws_alpha.send(r'{"op":"publish","topic":"/test_rosbridge_leak","msg":{"data": "foo, and also bar"}}')
    
        print ws_bravo.recv()
    
        ws_alpha.close()
        ws_bravo.close()
        # Simulate a rude socket abort:
        #ws_alpha.sock.shutdown(socket.SHUT_RDWR)
        #ws_bravo.sock.shutdown(socket.SHUT_RDWR)
    
    def run_cycles(max_cycles):
        for i in range(max_cycles):
            test_cycle(i)
    
    if __name__ == '__main__':
        run_cycles(NUM_CYCLES)

To reproduce,

* Start rosbridge: `rosrun rosbridge_server rosbridge_websocket`

* Check the number of child processes with `ps -eLf | grep rosbridge_websocket`

* Run the test script.

* Check the number of child processes again.  As of 0.7.13, there will be roughly one hanging child process per test cycle.

My proposed solution is to modify MultiPublisher/MultiSubscriber to, instead of calling `.unregister()` on the wrapped rospy.Publisher/rospy.Subscriber, delete it and *immediately* invoke destructors.  The above test results in no warnings and no leaked children.  Making a PR for that.

mvollrath, 2015-09-17:
This was a [leak in rospy QueuedConnection](https://github.com/ros/ros_comm/pull/661).  The Thread persists a reference back to `TCPRosTransport` which I saw piling up using objgraph.  With that PR there are no junk Threads or hanging registrations.  According to objgraph, no objects are consistently growing between sessions when I grind on the above test script.

There is still the race connect warning, but an interesting clue is that it goes away when switching the test from a graceful websocket close() to a lower level socket abort (`shutdown(socket.SHUT_RDWR)`)

mvollrath, 2015-09-18:
A point about the "not a publisher" warnings.  If you force garbage collection after removing a MultiPublisher, it will *always* show a warning.  i.e. `PublisherManager.unregister()`

```
        if not self._publishers[topic].has_clients():
            self._publishers[topic].unregister()
            del self._publishers[topic]
            import gc
            gc.collect()  # always produces "is not a publisher of" warning
```

This suggests that a value is being dereferenced (set to `None`) where it should be deleted.

mvollrath, 2015-09-18:
https://github.com/ros/ros_comm/commit/078292e11abce5a23f69517b88b355d9d94040dc fixes all rosbridge pubsub leaks as far as I can tell.  The warnings are still annoying.

closed 2021-09-19 by github-actions[bot]