RobotWebTools/rosbridge_suite #169: Websocket should use BSON to properly support PointClouds and other binary data

pbeeson, 2015-02-06:
I was using the rosbridge websocket server for a project, but we were sending LARGE amounts of point cloud data to a web page to be rendered.  It turns out of course that there were several issues.  The first of which is the fact that the roslibjs client side does keep up and memory leaks occur.  This was easily fixed by setting up a service to request large PC data via pulling instead of attaching to a broadcasting stream of hundread of MB of data a second.

However, we quickly noticed that in roslibjs, the majority of the time taken to render the PoinCloud was in decoding the base64 encoding of the binary data.  After looking into this, it turns out that with VERY minimal changes, rosbridge and roslibjs can be set up to use BSON (just 'apt-get install python-bson' and get the bson.js) and point clouds can be passed as binary blobs with no problem.  We've been running this for weeks and see a speedup from 65ms to unpack and render a 3D point cloud from a Carnegie Robotics Multisense to a 15ms.  That's a HUGE performance increase.

Below I tack on the siff to show how simplt my changes are, and would like to encourage rosbridge and roslibjs devels to switch to using BSON for server to client message passing, while still using JSON for client to server message, service, param comms as that is typically very small amounts of data.

```
pbeeson@tardis:~/projects/PROWL/code/rosworkspace$ diff -r rosbridge_suite/src ~/Desktop/rosbridge_suite-develop | grep -v pyc
Only in rosbridge_suite/src: CMakeLists.txt
diff -r rosbridge_suite/src/rosbridge_library/src/rosbridge_library/internal/message_conversion.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py
43d42
< import bson
104c103
<             return bson.Binary(inst)
---
>             return standard_b64encode(inst)
diff -r rosbridge_suite/src/rosbridge_library/src/rosbridge_library/protocol.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_library/src/rosbridge_library/protocol.py
42d41
< import bson
264,265c263
<             #return json.dumps(msg)
<             return bson.BSON.encode(msg)
---
>             return json.dumps(msg)
diff -r rosbridge_suite/src/rosbridge_server/scripts/rosbridge_websocket /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_server/scripts/rosbridge_websocket
105c105
<         IOLoop.instance().add_callback(partial(self.write_message, message, True))
---
>         IOLoop.instance().add_callback(partial(self.write_message, message))
diff -r rosbridge_suite/src/rosbridge_server/scripts/rosbridge_websocket.py /home/pbeeson/Desktop/rosbridge_suite-develop/rosbridge_server/scripts/rosbridge_websocket.py
105c105
<         IOLoop.instance().add_callback(partial(self.write_message, message, True))
---
>         IOLoop.instance().add_callback(partial(self.write_message, message))
```

And for roslib.js
```
pbeeson@tardis:~/projects/PROWL/code/BrowserUI/modified$ diff roslib.js ~/Desktop/roslib.js 
1,2d0
< var BSON = bson().BSON;
< 
1092,1102c1090,1095
< 	var reader    = new FileReader();
< 	reader.onload  = function() {
< 	    var uint8Array = new Uint8Array(this.result);
< 	    var msg = BSON.deserialize(uint8Array);
<   	    if (msg.op === 'png') {
< 		decompressPng(msg, handleMessage);
< 	    } else {
< 		handleMessage(msg);
< 	    }
< 	}
< 	reader.readAsArrayBuffer(message.data);
---
>       var data = JSON.parse(typeof message === 'string' ? message : message.data);
>       if (data.op === 'png') {
>         decompressPng(data, handleMessage);
>       } else {
>         handleMessage(data);
>       }
```

pbeeson, 2015-02-06:
Also added this as an issue to roslibjs as it affects both server and client.
https://github.com/RobotWebTools/roslibjs/issues/155

rctoris, 2015-02-06:
Have you looked at the point cloud encoder and streaming solution provided in ros3djs? Data is streamed in a very efficient manner for rendering by encoding the point cloud inside a compressed video stream: https://github.com/RobotWebTools/ros3djs/blob/develop/examples/depthcloud.html

In general, I feel JSON is essential to the ideas behind rosbridge including ease of use and availability of client libraries.

pbeeson, 2015-02-06:
Yeah, we had multiple issues with the depth cloud encoding.  First, it seemed to have hard coded stuff in it that worked great for Kinect/Asus Xtion, but not for generic pointcloud sources (i.e., some that might encode RGB/termperature/luminance as floats or mono16, and for really dense point clouds from actaul stereo devices that aren't limited to 4m and 640x480).  

Also, BSON just sits on top of JSON, providing a Binary Blob type and binary encoding of the data.  It's been out for a long time, and is working great.  My diff above shows how minimal the changes need to be.

To me, it seems like a bad idea to have a special PNG chucker/delivery for images, a special compressed video streamer for depth images, the base64 for all other binary data, when you can just send binary arrays a blobs with minimal overhead. If you look at my "patches" above, you'll see the minimal amount of changes this requires to roslibjs and rosbridge_server.  I'm not saying this has to replace JSON --- you could keep the base64 encoding/JSON for legacy issues and just make this a runtime parameter, where if people want to send ASCII JSON arrays in a bloated, slow way they can, but if they want binary data to stay binary, they have that option.

Just trying to offer some alternatives for users.  We prefer getting the raw, unaltered data and prefer not to have to spend valuable time decoding a 50MB base64 message in a JS display program.

rctoris, 2015-02-06:
Perhaps instead of defaulting to BSON for subscriptions (which would break additional clients other than the roslibjs client), it could be a new type of field for the `compression` type (https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe). That way, a client who supports BSON could specify the topic be transmitted this way and rosbridge could send it appropriately.

pbeeson, 2015-02-06:
That seems reasonable to me.

rctoris, 2015-02-06:
Assuming no objects to this (I think this sounds like a great addition), could you put together a PR with your proposed changes?

rctoris, 2015-02-10:
Could also address https://github.com/RobotWebTools/roslibjs/issues/36

DLu, 2015-06-26:
I have made some progress on this front based on @pbeeson 's work. 

## Changes to rosbridge
https://github.com/DLu/rosbridge_suite/commit/d7138dc810aa77b8066a0a2d77195172221a1977
Similar to @pbeeson but this only sends in binary when the message contains binary information (uint8[] or char[])

## Changes to roslibjs
https://github.com/DLu/roslibjs/commit/cbc023473fd65d7def5650d15e113f2d107f5082
Backwards compatible to receive binary and json messages. 

Limitation: Requires you to also manually include bson.js in the HTML. 
@pbeeson : How did you include this in your html/javascript? 

## Changes to ros3djs
https://github.com/RobotWebTools/ros3djs/compare/develop...DLu:pointclouds
Here is my PointCloud2 display module which, unlike the DepthCloud module can handle general pointclouds that are not image-like. 
[demo](http://youtu.be/wEu1DHAHlh8)

This method for PointCloud display does not require running the additional webserver on an alternate port to send the video stream. However, the ros3djs code assumes the binary representation (using the above patches). I tried quickly to make a non-binary version but couldn't figure out how to unpack the bits from the string.

T045T, 2015-07-25:
Can we close this, now that #190 is merged?

pbeeson, 2015-07-26:
I haven't checked this against my code but I hope to get a chance in the
next few weeks.

Feel free to close.

closed 2015-07-27 by rctoris

kshitijgoel007, 2016-08-20:
Hello Everyone,

Pardon me for posting on a closed issue, but I'm in a similar situation. I'm transferring a custom made message composed of Stochastic params (extremely compressed Dense Point Clouds). It is **lower in size than sensor_msgs/PointCloud**, but is _significantly high_ to such extent that I am **observing high network latency**. Moreover, I am **not reading data using roslibjs**. I am using [this](https://github.com/michaeljenkin/unityros) repo in Unity C# to parse data. Currently, I have a nicely working JSON parser which parses my data with some latency.

Will this BSON enhancement work for such custom messages too? Other than writing a BSON parser, what other enhancements would my repo require?

Thanks for having a look. Awaiting your response.

Kshitij

rcodddow, 2016-08-20:
Hi I am a contributor to the ROSBridge Unity repo you are using and I can tell you that BSON messages will not work with the code in its current form. There are a number of modifications that  you will have to make aside from a BSON parser. Please post you questions here https://github.com/michaeljenkin/unityros/issues

kshitijgoel007, 2016-08-25:
Hello @rcodddow , please have a look at https://github.com/michaeljenkin/unityros/issues/4. Just need a few ideas.