RobotWebTools/rosbridge_suite #171: Websocket handler and Nagle's algorithm

smamessier, 2015-02-16:
Using Rosbridge Websocket and Roslibjs, I subscribe to a 60 Hz Ros message at a 15 ms throttle rate with 0 queue size, I end up receiving 6 messages at a time but every 100 ms.
I took me some time but I tracked this lag down to Nagle's algorithm that gathers tcp messages together before sending when small and sent at a high frequency. 
This way, low throttle rate (typically under 50 ms) are only respected in average and this is a huge problem when you want some decent head tracking for instance.
 
Using Tornado WebSocketHandler API, I tried tu use self.set_nodelay(True) which is supposed to deactive Nagle but without success. (Still the same message scheduling pattern)

I only found a kludgy way of achieving 15 ms between each message by forcing Roslibjs to ping back the websocket server on each message reception (this forces Nagle's algorithm to flush). Idea come from this post 
http://stackoverflow.com/questions/13373519/websocket-tcp-packets-clumping-together/13406438#13406438

smamessier, 2015-02-25:
I had the opportunity to test without a proxy, and adding 
    self.set_nodelay(True)
in rosbridge_websocket.py  line 63 (opening connection callback), fixes the issue. This should either be added or available as an option.

jihoonl, 2015-02-26:
@smamessier Would you create a PR for your patch?

smamessier, 2015-02-27:
Ok, I created a PR here #175

T045T, 2015-03-15:
Closed via #175

closed 2015-03-15 by T045T