RobotWebTools/rosbridge_suite #181: Node doesn't shutdown properly when another node registers

BennyRe, 2015-05-11:
When the node is killed with Ctrl+C the node exits completely.

When the node is killed because another node with the same name is started (launching the rosbridge_websocket launch file a second time) the node doesn't close the socket. It stays in a semi-alive state. New connections are accepted but they don't work the way the normally do. The new node is not able to start because it can't create the socket.

    socket.error: [Errno 98] Address already in use

Trying to launch it again leads to the same error.

Ctrl+C'ing the first node closes the socket and a new rosbridge can be launched.

Is this intended behavior or a bug?

This is the output when another node registers:

```
$ roslaunch rosbridge_server rosbridge_websocket.launch 
... logging to /home/iki/.ros/log/c01d1998-f7ba-11e4-aa7c-00012957991b/roslaunch-marvin-3009.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.5.2:51827/

SUMMARY
========

PARAMETERS
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/port: 9090
 * /rosdistro: indigo
 * /rosversion: 1.11.10

NODES
  /
    rosapi (rosapi/rosapi)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://192.168.5.2:11311

core service [/rosout] found
process[rosbridge_websocket-1]: started with pid [3041]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.stop_service.StopService
process[rosapi-2]: started with pid [3068]
[INFO] [WallTime: 1431342306.728797] Rosbridge WebSocket server started on port 9090
shutdown request: new node registered with same name
[rosapi-2] process has finished cleanly
log file: /home/iki/.ros/log/c01d1998-f7ba-11e4-aa7c-00012957991b/rosapi-2*.log
```

BennyRe, 2015-05-19:
Do you guys don't have this behavior?

T045T, 2015-05-22:
Fixed in #182

closed 2015-05-22 by T045T