RobotWebTools/rosbridge_suite #191: Sending message from custom Ground Control Station to rosbridge server using TCP link problem.

MauroPagliarella, 2015-08-01:
Hi all
I'm building a robot running ROS on board and I would like to send commands to it using custom ground control station I'm developing by Qt.

SYSTEM CONFIGURATION:
1: Laptop running Ubuntu 14.04 and my custom Qt ground control station
2: Raspberry PI running ROS robot control node and rosbridge_server TCP node

OBJECTIVE:
Connect the ground Control Station to rosbridge TCP server and send std_msgs/Int32MultiArray messages to the robot control node.

OPERATIONS:
1: roscore started on raspberry terminal
2: rosbridge started on another raspberry terminal (command: roslaunch rosbridge_server rosbridge_tcp.launch) 
3: Qt ground control station started, connected to rosbridge server and sent the following advertise message to rosbridge
    "op"       = "advertise";
    "topic"    = "/TcMsg";
    "type"     = "std_msgs/Int32MultiArray";
4: rostopic echo started on another raspberry terminal (command: rostopic echo /TcMsg)
5: Sent JSON message to publish over /TcMsg topic
    My Qt code for this operation is:
    ---
    QString Msg;
    Msg.clear();
    Msg.append("{\"msg\" : \"data\" : [");
    tmp = QString("%1, ").arg(1, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(2, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(3, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(4, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(5, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    //
    tmp = QString("%1, ").arg(6, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(7, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(8, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(9, 0, 10);    Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(10, 0, 10);   Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(11, 0, 10);   Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(12, 0, 10);   Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(13, 0, 10);   Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(14, 0, 10);   Msg.append(tmp); Msg.append(" , ");
    tmp = QString("%1, ").arg(15, 0, 10);   Msg.append(tmp); Msg.append("], ");
    Msg.append("{\"layout\": {\"dim\": [], \"data_offset\": 0} , \"topic\" : \"/TcMsg\", \"op\" : \"publish\" }");
    Msg.append("}");

    QByteArray Tc = QByteArray(Msg.toUtf8());
    if(TCP_TcLinkSocket->write(Tc.data()) >= 0) {
        if(!TCP_TcLinkSocket->flush()) {
            QMessageBox::information(this, tr("TCP client socket flush FAILED!"), tr(""));
        }
    }
    ---

RESULTS:
Operation 1 seems OK

Operation 2 seems OK and
From rosbridge terminal I got the following output messages:
...
...
[INFO] [WallTime: 1438455965.531628] [Client 1] connected. 1 client total.

Operation 3 seems OK and I got no output message from rosbridge

Operation 4 seems OK and I got the following output messages after the execution of operation 5:
pi@raspberrypi ~/catkin_pi $ rostopic echo /TcMsg 
layout: 
  dim: []
  data_offset: 0
data: []
---

Moreover, the callback on receiving /TcMsg is started but seems not to get data coherently to rostopic echo /TcMsg output.

Probably there is an error on the JSON message I sent. I tried to modify it but every time got the rosbridge error: Received a message without an op. ...

How can I send a std_msgs/Int32MultiArray to rosbridge so that it publishes my message to my robot node topic?

Could somebody help me please?
Thank you very much in advance

MauroPagliarella, 2015-08-02:
In the end my attemps succeeded. The correct string to send to rosbridge is:
{\"op\" : \"publish\", \"topic\" : \"/TcMsg\", \"msg\" : {\"layout\": {\"dim\": [], \"data_offset\": 0}, \"data\":[1, 2, 3, 4, 5]}}

nmbferreira, 2017-08-09:
@MauroPagliarella hi, I'm developing an android app to talk with my computer's roscore and, thanks to you, i managed to send messages to topics via JSON. The next thing i tried was to receive messages from ros, trying to subscribe to a topic, but was unsuccessful. Any tips on how to do it? Thanks in advance!

Hate95, 2020-04-30:
@MauroPagliarella 
Hello, could you share the source for communication on this protocol?

closed 2021-09-18 by github-actions[bot]