RobotWebTools/rosbridge_suite #196: Error processing request when calling rosapi/service_type: field type must be of type str

ric-mendonca, 2015-10-27:
Hi all, 
after launching roslaunch rosbridge_server rosbridge_websocket.launch on a terminal, if I run rosservice call /rosapi/service_type "service: '/rosapi/topics'" on a new terminal, the rosbridge_server will output the following error:

[ERROR] [WallTime: 1445944939.168494] Error processing request: field type must be of type str

When I traced back the error I found that the culprit lies in Proxy.py of rosapi package. Specifically, the proxy's function "get_service_type" should not be calling rosservice_find() but instead rosservice_get_service_type(). The former expects a string variable with an already defined service type, but what is in fact in that variable is the service name (topic). This will lead to the mentioned error.

Please consider correcting this bug in rosbridge_server :)

Proposed solution:

File Proxy.py at Line 86:

def get_service_type(service):
    """ Returns the type of the specified ROS service, """
    try:
        return rosservice_get_service_type(service) #rosservice_find(service)
    except:
        return ""

T045T, 2015-11-02:
Hi!

Thanks for reporting this, and especially for going the extra mile and tracking down the origin!
If it isn't too much trouble, could you make a pull request with your fix? It seems only fair to put your name on that commit :)
Also, a test to a) confirm your change fixes the error and b) guard against regressions would be great.

ric-mendonca, 2015-11-05:
Hi T045T, 

I noticed now that the user rcodddow already found and pull requested this fix a month ago!  :)

closed 2015-11-05 by ric-mendonca