RobotWebTools/rosbridge_suite #198: Rosbridge only one way on OSX

cfry, 2015-11-25:
This issue is more fully described by:
http://answers.ros.org/question/221356/rosbridge-only-one-way-on-my-osx/

In a nutshell, I can run the tutorial http://wiki.ros.org/roslibjs/Tutorials/BasicRosFunctionality
with messages successfully leaving my browser, reaching the tutorial server thru rosbridge,
and successfully computing the answer (according to printouts in the terminal window) , but nothing comes back to the browser.  (the callback isn't called and doesn't look like any web  messages
are coming in either according to Chrome Dev Tools.)
No errors occur that I can see. 
I can run the turtlesim demo fine and overall ROS seems to be behaving well other than this problem.

From the above ROS answers URL, you can see that commenter Philipp was able to run this
tutorial successfully on his Ubuntu machine and that he had similar 
ROS environment variables and /etc/hosts as me. 
Answerer William Woodall was able to reproduce this problem on his OSX Yosemite & Indigo machine
(same as mine) and suggested I open this issue in rosbridge. 
Comments in his answer sound like good clues to start with.

NOTE I would be happy if this issue was already fixed in Jade, but I have not tested it in Jade as
installing ROS on OSX is, um, difficult for me, but worth the switch if that's what it takes to resolve
this as its a showstopper for my app.

PS I'm running Python 2.7 and downloaded ROS indigo and Rosbridge about 2 weeks ago.
THANK YOU!

T045T, 2015-11-25:
I'll try to get Indigo set up on my MacBook and reproduce this for testing by this weekend - sorry, I won't get to it any earlier. Any others with OS X machines?

A couple of questions:
* Are you using the system Python or a brewed version?
* In your Answers post, you mention that rosbridge was difficult to install - can you write down the steps you took to install it? Maybe they'll get us started on brew support for rosbridge :)

jonfink, 2015-11-25:
Also having this same issue! 
- Using system python
- rosbridge install was smooth for me (cloned rosauth and rosbridge_suite git repos into a workspace)

I've started to dig in to see where messages (published from ROS) stop moving forward but I've only gotten so far as to determine that `Protocol.send` is being called but the output of `self.serialize` is `None`

jonfink, 2015-11-25:
bson dependency turned out to be the issue. Doing a `pip insatll pymongo` (which includes it's own bson module) and things seem to work for me now.

cfry, 2015-11-26:
Three Cheers! 
pip install pymongo
solved  the problem.
T045T asks what version of python am I running.
I'm aware that the Mac has its own and ROS has its own but
am confused about how to find out which one is running etc.
How about extending rosversion by having it output lots
of info like the python version, maybe the ROS environment variables, the OS,
and anything else that someone who was trying to fix a bug would 
want to know.
I installed rosbridge by:
- from  https://github.com/RobotWebTools/rosbridge_suite
  downloading the zip, expanding and moved it under ros_ws/src/
- ran catkin_make 
- ran  roslaunch rosbridge_server rosbridge_websocket.launch
- which errored with: ImportError: No module named rosauth.srv
-From  https://github.com/WPI-RAIL/rosauth, downloaded the zip, expanding
  and moving upder ros_ws/src/
- Renamed the folder to "rosauth"
- catkin_make   and launched rosbridge again to discover
 that it needed BSON.
- pip install bson
- then somehow figured out I needed twisted so
- pip install twisted
-Then launching rosbridge didn't error but of course didn't work either.
So replace pip install bson above with pip install pymongo
and I don't know if pip install twisted is still necessary.

Yes of course a brew rosbridge would be way better,
but better still would be just included rosbridge in the main ROS install.
I am concerned with bloat of unnecessary packages too but:
You've already got rospy_tutorial in there and maybe a few other
unnecessary things so I'd say make the main ros install have lots
of features that are good for students, etc.
and then have a separate bare-bones ros for people that are deploying it
on a small processor, as presumably they know more about what they're
doing than the noobs like me will.

Now going a bit deeper into solving problems:
The ros tutorials are nicely formatted, the problem is that the
content isn't complete enough for us idiots out here in userland.
So for instance: jonfink says in one of his comments:
"cloned rosauth and rosbridge_suite git repos into a workspace)"
its exactly these kinds of comments that are "incomplete" in a tutorial
and the tutorials have all kinds of assumptions of what the users know
in them. On the other hand, johnfink also says
"pip insatll pymongo"
which is much better. It tells me HOW to do something not just what to do.
Now the problem with that is that "insatll" is misspelled. 
OK not a problem for me but its nice if we could have a general way to
avoid these kinds of problems in tutorial instructions.

The general way is to user test each tutorial, but that's too time consuming
for any tutorial writer to do. But suppose that there's a special markup
for "a step" in a tutorial. And each one of those has beside it
a clickable icon, say "?". Now clicking means you can ask a question
about that specific step. Clicking on the "?" pops up an email message
<a href="mailto@the-tutorial-maintailer?subject=definitive-tutorial-name?body
tutorial-step-number_and_debug_info>?</a>
OK now if we can inject into the body of that email message not just
which step of the tutorial (or maybe the step text)
as well as things the tutorial maintainer might want to know, like
the python version, ROS env vars, ROS version, OS & version, etc.
then the tutorial maintainer can get a decent bug report even if the
hapless user doesn't know enough to make a decent bug report.
(thus avoiding email round trips of developers asking which version of python, etc.)
Note that to get the rosversion and the py version etc, use my extended
rosversion cmd above with rosbridge connecting to ROS to automatically get 
this info. (another reason to have rosbridge always installed!)
[but in the mean time, you can just TELL the user to call rosversion
and roswtf, and paste the result into the email.]

There are several categories of responses to the user that should be made,
all of them short and none of them telling them (exactly how) to fix the problem.
- If its a real bug: tell them to download the new fixed software
- If the doc was wrong or confusing, tell them to read the now updated doc
- If its a silly question, no need to respond at all.
If a tutorial maintainer has the discipline to fix confusing/wrong doc
first then respond, they'll be rewarded with no similar reports and only
explaning what to do exactly once.

Thanks for the speedy response, guys!

cfry, 2015-11-26:
&lt; a href="mailto@the-tutorial-maintailer?subject=definitive-tutorial-name?body
tutorial-step-number_and_debug_info&gt;?&lt;/a&gt;
OK now if we can inject into the body of that email message not just
which step of the tutorial (or maybe the step text)
as well as things the tutorial maintainer might want to know, like
the python version, ROS env vars, ROS version, OS & version, etc.
then the tutorial maintainer can get a decent bug report even if the
hapless user doesn't know enough to make a decent bug report.
(thus avoiding email round trips of developers asking which version of python, etc.)
Note that to get the rosversion and the py version etc, use my extended
rosversion cmd above with rosbridge connecting to ROS to automatically get 
this info. (another reason to have rosbridge always installed!)
[but in the mean time, you can just TELL the user to call rosversion
and roswtf, and paste the result into the email.]

There are several categories of responses to the user that should be made,
all of them short and none of them telling them (exactly how) to fix the problem.
- If its a real bug: tell them to download the new fixed software
- If the doc was wrong or confusing, tell them to read the now updated doc
- If its a silly question, no need to respond at all.
If a tutorial maintainer has the discipline to fix confusing/wrong doc
first then respond, they'll be rewarded with no similar reports and only
explaning what to do exactly once.

Thanks for the speedy response, guys!

T045T, 2015-11-26:
Thanks for the input!
While there is certainly something to be said for improving the tutorials as much as we can, let's take a step back:
You say you're in "userland", but (as much as it pains me to say this) running ROS on OS X and checking out git repositories are both not exactly user-y stuff.
Normally, you'd use a package manager to install a packaged version and resolve the resulting dependencies. In your case, the way to resolve the BSON dependency would have been to use [rosdep](http://wiki.ros.org/ROS/Tutorials/rosdep), like so:
```
rosdep install rosbridge_server
```
At least that would be nice, but there doesn't seem to be a rosdep rule for ```python-bson``` on OS X (see [the rosdistro list](https://github.com/ros/rosdistro/blob/46ac4888c353b66a0dbf44993b43ef907a05ff09/rosdep/python.yaml#L228)).
Regardless, I feel like maybe ```rosdep``` doesn't get mentioned in the tutorials as much as it ought to, considering how important it is to have all the dependencies for your packages. On the other hand, it could just be me not looking at the documentation until recently.

Of course, even after manually installing the dependencies, there still seems to be an issue - you've come to the right place in opening this issue! The Github URLs being prominently featured on the ROS wiki pages (and the need for you to clone it in the first place) hopefully made it easy for you to find us.

As for advice on tutorials, ROS Answers seems a better place to ask, because it allows more people than just the maintainers to provide their knowledge.

T045T, 2015-11-26:
@jonfink Our ```package.xml``` lists ```python-bson``` as a dependency.
There isn't a rosdep rule to install it on OS X yet, but on Ubuntu, it resolves to installing the ```python-bson``` package using ```apt-get```.
That's the BSON implementation from MongoDB, probably the same you installed with ```pip install pymongo```.
Do you know a way to install just the ```bson``` module, or is installing all of ```pymongo``` the only way?

T045T, 2015-11-26:
Note: [MongoDB documentation](http://api.mongodb.org/python/current/installation.html) explicitly warns agains doing ```pip install bson``` because of the exact compatibility problem we've encountered.

jonfink, 2015-11-29:
While I _should_ use `rosdep install ...`, I tend not to due to rosdep rules being unfilled, out of date, or doing things I don't want them to on OSX (e.g., install homebrew python in the past...). The difference between MonoDB BSON and `pip install bson` is devious and tough to track down when the dependency for rosbridge_suite is listed as BSON. But I suppose this is more a result of Ubuntu package naming (python-bson points to a piecemeal install of the BSON component of MongoDB) than anything else.

wintermuteger, 2016-07-24:
I had the same issue with my ROS installation using meta-ros. Basically I had to manually install (no catkin on target platform) BSON. Unfortunately I used also the "standard" and not the mongodb version.

If someone reads this facing a similar issue without access to pip or ros installers on target systems: Just copy the bson directory from the python-mongodb library to /opt/ros/<distro>/lib/python2.7/site-packages. Problem solved afterwards.

As this seems to be an issue that already half a dozen people faced, and it is tricky to debug (one direction works, so connection is established, no error displayed, just no response received) my proposal/wish would be, to check if the BSON module supports e.g. Binary and if not display at least a warning or even an error stating: "BSON installation does not support all necessary features. Please use the MongoDB BSON implementation." This would have saved me quite some hours (and as I am working only in my off hours on it, maybe even some month), and might also help some others avoiding this pitfall.

I would add it myself, unfortunately I am lacking the necessary Python skills :-(

ledmonster, 2017-03-10:
Created a pull request to check whether appropriate bson module is installed or not. I hope it would improve current situation.

subvertallchris, 2017-11-16:
Thanks for everyone's work on this. Had and fixed the same issue.

jihoonl, 2017-11-20:
change in #270 will raise exception if a proper bson module as of now.

closed 2017-11-20 by jihoonl

laghbani, 2020-05-10:
do this:

# sudo pip uninstall bson

# python -m pip install pymongo