RobotWebTools/rosbridge_suite #199: How can I get kinect image data via TCP connection?

chenxian0723, 2015-12-02:
I want to gain image data produced by kinect on Turtlebot2, and I see that the image data topic is /camera/rgb/image_color.
Now I can get image data on Turtlebot2 and save to a ".jpg" file using opencv2(python), but when I try to subscribe the topic via the TCP method of ros_bridge, I coundn't get the full data. Why? 

My message is " {'op':'subscribe', 'topic':'/camera/rgb/image_color'} "
And I use: 
while Ture:
    tmp = sock.recv(1024)
    if len(tmp) != 1024:
        break

rcodddow, 2015-12-02:
The topic name is not important here only the topic type. 

Please publish the entire source (indented properly )as python is weakly typed and white space dependant. Your snippet provides very little information. You also spelt True wrong in your while loop. Additionally most tcp implementations of recv(<some int>) return the number of byte read (usually less than <some int>) and you need to continually poll them until you have read the required amount of bytes. It looks as if you have coded a naive implementation as 1024 bytes for a colour image from a kinect is very low.

chenxian0723, 2015-12-02:
I tried to poll the data several times and print them out, but could not get the full formatted message (see the attach files).
I expected to get the tail of data as " XXXX}, width='640'} ", but failed.   
[image]
[image]

rcodddow, 2015-12-02:
this is data from a sensor_msgs/Image which is raw camera data formatted as bgr8. You need to read more that 1024 bytes. Please look at the rosbridge protocol for transmitting messages. You can't determine the number of byte you need to read in advance. It looks like you will need to read until the next newline character "\n" and then parse the entire message.

rcodddow, 2015-12-02:
please post all relevant source code.

chenxian0723, 2015-12-02:
It seems the buffsize must be given when using recv().  
Is there any simple demos of receiving image using ros_bridge?

And this is my source code:
[1.txt](https://github.com/RobotWebTools/rosbridge_suite/files/49598/1.txt)

rcodddow, 2015-12-02:
The tcp protocol in rosbridge is relativly new, as to my knowledge there are no tutorials or examples on its use. There are also very few examples on using rosbridge in any particular programming language as it is a bridge from ros to anywhere.

For info on the socket module in python please refer to its documentation (https://docs.python.org/2/howto/sockets.html) and ask google the rest.

pttp, 2015-12-02:
Please have a look at this:
https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/test/experimental/fragmentation%2Bsrv%2Btcp/test_non-ros_service_client_fragmented.py

yghcats, 2017-08-29:
@chenxian0723 
Hello, how do you change your PNG pictures?
I don't know how to convert the laser point cloud data I'm currently getting

Thank you！

closed 2021-09-18 by github-actions[bot]