RobotWebTools/rosbridge_suite #214: Release rosbridge_suite for kinetic

cberner, 2016-04-02:
Will rosbridge_suite be released for kinetic? I'm trying to port my library (rospilot) to kinetic, but have a dependency on rosbridge_suite

rctoris, 2016-04-09:
Will put this on the todo list, thanks!

doisyg, 2016-06-13:
+1 for kinetic repo release.
Compiling from source works fine but requires installing the dependencies and compiling the ones that are not yet released for kinetic (rosauth to this date)

TinyTheBrontosaurus, 2016-06-15:
+1 here as well.

@doisyg (or anyone) I've only ever used apt-get in the past for ROS. Can you point me at instructions for installing from source?

doisyg, 2016-06-15:
1 git clone the desired package in your ros workspace/src. Example: 
```
cd ~/catkin_ws/src
git clone https://github.com/RobotWebTools/rosbridge_suite.git
```

2 build your workspace:
```
cd ~/catkin_ws
catkin_make
```

jihoonl, 2016-06-22:
Didn't find the clean solution to separate out tornado as library yet from rosbridge_server. As soon as we resolve #162, I can release rosbridge in kinetic.

jihoonl, 2016-06-27:
Here is the plan to release rosbridge in kinetic.

1. Final version release in indigo and jade with latest changes.
2. Create a new branch, and try to resolve #162 as suggested by https://github.com/RobotWebTools/rosbridge_suite/issues/162#issuecomment-220920725. 
3. No matter it is successful or not, i will release it before August.

jihoonl, 2016-06-27:
If anyone helps me on this, it can speed up.. :)

jihoonl, 2016-08-08:
Well.. I couldn't find any clean way of separating out Tornado from rosbridge_server. I will keep the current structure until someone helps me on #162.

I will release on Kinetic after merging #233.

asmodehn, 2016-08-12:
@jihoonl Lets catchup sometime and we can solve #162 together.

It should now be easy enough to make a ros package as third party with catkin_pip.
Example with click : https://github.com/asmodehn/click-rosrelease

We just need to decide on a tornado version that works for all tornado related projects, for each ROS distro, and I can guide you to do the third party releases (or I can do them myself, here, or fork, just let me know).

jihoonl, 2016-08-15:
Kinetic rosbridge is rolling now with old structure.

@asmodehn I have created experimental-tornado branch where we can continue experiments. I will follow up the status in #162

jihoonl, 2016-08-18:
Done.

closed 2016-08-18 by jihoonl