RobotWebTools/rosbridge_suite #218: Untracked run dependency: Twisted

mvollrath, 2016-04-25:
The UDP handler requires Twisted.  This breaks all the other server scripts (tcp, websocket) if it's not available, because it is imported in the module init:

https://github.com/RobotWebTools/rosbridge_suite/blob/a4a407ae8afe845eec0522efa4cecadccfa009de/rosbridge_server/src/rosbridge_server/__init__.py#L3

Which then raises an `ImportError` when the UDP handler tries to import Twisted:

https://github.com/RobotWebTools/rosbridge_suite/blob/a4a407ae8afe845eec0522efa4cecadccfa009de/rosbridge_server/src/rosbridge_server/udp_handler.py#L6

`ImportError: No module named twisted.internet.protocol`

Run dependencies need to be tracked in package.xml.  In Indigo on Ubuntu trusty, the `python-twisted-core` package is sufficient to suppress this error, but I can't say whether or not the UDP handler will work against this version of Twisted.

Fortunately `python-twisted-core` is in the rosdep tree and this is a one-line fix:

https://github.com/ros/rosdistro/blob/17fa310173bc4cd4a9b240f44f9fe3f976dba777/rosdep/python.yaml#L2209-L2221

closed 2016-04-25 by rctoris

ruffsl, 2016-05-25:
Has this been released into indigo yet?

mvollrath, 2016-05-27:
0.7.14-1 is now published, but it does not include the new dependency.

```
$ apt-cache show ros-indigo-rosbridge-server
Package: ros-indigo-rosbridge-server
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 1615
Maintainer: Russell Toris <rctoris@wpi.edu>
Architecture: amd64
Version: 0.7.14-1trusty-20160425-152101-0700
Depends: libc6 (>= 2.2.5), ros-indigo-rosapi, ros-indigo-rosauth, ros-indigo-rosbridge-library, ros-indigo-rospy
Description: A WebSocket interface to rosbridge.
Description-md5: 6176459735b549c2c8121b7c7c9ac7aa
Homepage: http://ros.org/wiki/rosbridge_server
```

mikepurvis, 2016-06-27:
Ping for Indigo release.

jihoonl, 2016-06-28:
https://github.com/ros/rosdistro/pull/11995

PR #219 (2016-04-25): Track Twisted run_depend
Fixes #218
commit 6150234b (2016-04-25): Track Twisted run_depend

Fixes #218