RobotWebTools/rosbridge_suite #220: Can we have a todo-list for compatibility with kinetic ?

Kiloreux, 2016-04-26:
That's a question, can we please make a todo-list for all the changes  that need to be implemented in order for kinetic compatibility ?

rctoris, 2016-04-28:
I'm not sure there are any changes given http://wiki.ros.org/kinetic/Migration

It's a question of testing and making sure the necessary upstream repos are in (which they should be). I have no migrated to kinetic so I have not done any testing.

Kiloreux, 2016-04-28:
Cool, I will update this issue once I do the testing over the new kinetic with required changes.

cberner, 2016-05-24:
Looks like Kinetic was released today. Any update on this? Would it be helpful for me to do some testing?

dcconner, 2016-06-03:
Since it is not released, I built from source under kinetic.
I had to also build rosauth  as it is not released as package either, and needed to manually install python-bson, but otherwise it appeared to work for my code without modification.

Kiloreux, 2016-06-03:
@dcconner did you run all the tests ?

dcconner, 2016-06-03:
No just my use case.  Just providing a data point for others that want to
try

-- 
Best regards,

David C. Conner, PhD
Assistant Professor
Department of Physics, Computer Science and Engineering
Christopher Newport University
329 Luter Hall
1 Avenue of the Arts, Newport News, VA, 23606, USA
david.conner@cnu.edu
(757) 594-7360

dcconner, 2016-06-06:
I ran "rostest rosbridge_library test_all.test"

105 tests success; 0 errors; 0 failures

Kiloreux, 2016-06-06:
@dcconner , Thank you, closing issue for now.

closed 2016-06-06 by Kiloreux