RobotWebTools/rosbridge_suite #222: Security for rosbridge

AlexisTM, 2016-05-05:
Dear,

I am wondering about the security of ROSBridge. Is there any security protocol implemented or should we implement it via rosbridge_library ? 

By security protocol, I mean : User connection, rights management, user groups. 

NOTE : I am using rosbridge on private local network unconnected to internet for now, but it is not really a security solution.

rctoris, 2016-05-05:
robridge has the ability to use rosauth (https://github.com/GT-RAIL/rosauth) which does authentication. The general ideas behind it allow for authorized clients but currently does not have an open implementation of the user-role mechanism. Details are listed in the following paper: http://web.cs.wpi.edu/~cshue/research/tepra14.pdf

AlexisTM, 2016-05-05:
Thanks,

Rosauth seems to be very basic, therefore I might develop an user-role implementation when needed, but not right now.

closed 2016-05-05 by AlexisTM