RobotWebTools/rosbridge_suite #223: InvalidPackageException: Unable to load the manifest for package rosgraph_msgs.

manacoa, 2016-05-06:
Hi,

I'm using the rosbridge internal library to get load the message class for a given message type:

    from rosbridge_library import ros_loader
    ...
    self.msg_class = ros_loader.get_message_class(self.msg_type)

where self.msg_type is, for example, "rosgraph_msgs/Log".

This is generating the following error:

```
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 88, in get_message_class
    return _get_msg_class(typestring)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 128, in _get_msg_class
    return _get_class(typestring, "msg", _loaded_msgs, _msgs_lock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 163, in _get_class
    cls = _load_class(modname, subname, classname)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosbridge_library/internal/ros_loader.py", line 187, in _load_class
    raise InvalidPackageException(modname, exc)
InvalidPackageException: Unable to load the manifest for package rosgraph_msgs. Caused by: 'NoneType' object has no attribute 'strip'
<Greenlet at 0x7f1c3cc52230: <bound method RosGraphApi.launch_graph of <luxagent.ros_communication.ros_graph_api.RosGraphApi object at 0x7f1c3cc26090>>> failed with InvalidPackageException
```

This error is occurring on all message types, not just "rosgraph_msgs/Log".

Any suggestions would be appreciated.

Thanks!

Simon

manacoa, 2016-05-06:
I should have added that I'm running on Indigo.

jihoonl, 2016-05-06:
Please use [ros answers](http://answers.ros.org/questions/) for questions.

It looks like `rosgraph_msgs` is not detect in your ros package path. Make sure that correct ros environment is set in the PC.

jihoonl, 2016-05-07:
Now it is handled in rosanwers.

http://answers.ros.org/question/233758/rosbridge_suite-invalidpackageexception-unable-to-load-the-manifest-for-package-rosgraph_msgs/

closed 2016-05-07 by jihoonl