RobotWebTools/rosbridge_suite #239: Long/random time period before published messages arrive

follesoe, 2016-10-20:
If my rosbridge client reconnects and starts publishing messages to a topic, it takes a random amount of time before the messages suddenly start appearing.

We test this by using ``rostopic echo /my_topic``, and observe that the messages are arriving. Then we refresh the web client, which creates a new connection and start publishing messages to the same topic. The amount of time it takes from the client starts publishing messages and the echo-command displaying them seams a bit random. From a couple of seconds up to 30-40 seconds (and sometimes even more).

If we look at the diagnostics, rosbridge is outputting messages saying that the connection is open - so the web socket connection is established, and the client is publishing messages.

Any tips on how to debug/configure/fix this?

kjeremy, 2017-01-13:
I'm seeing this too with the node example.

robobit, 2017-03-16:
Same problem here, any progress or ideas how to troubleshoot it?

kjeremy, 2017-03-16:
This basically makes this project unusable.

T045T, 2017-03-17:
I just tried to reproduce this with the latest version of rosbridge and did not experience any problems.

I think this might be related to #138 and thus should be fixed with #247 - can you report back on whether your use cases still break?

kjeremy, 2017-03-18:
Stupid question but how do I install the latest version? There doesn't seem
to be a package for it on Ubuntu 14.04 ROS indigo.

ablakey, 2017-03-18:
My understanding is that they don't get built as regularly, so if you want the latest package, you'll need to clone it and build from source.

http://wiki.ros.org/catkin/Tutorials/using_a_workspace

T045T, 2017-03-18:
Looks like the latest releases for Indigo and Jade are pretty old by now so yes, building from source is your safest bet, and should be fairly straightforward - just don't forget to use rosdep to install all system dependencies :)

kjeremy, 2017-03-21:
It seems to work on the latest.

robobit, 2017-03-31:
I've build the latest version from source on 16.04/Kinetic but this doesn't seem to fix it for me.

T045T, 2017-03-31:
Can you share a self-contained test case that reliably fails? I tried, and couldn't create one :/

sjlevine, 2017-06-20:
Hello! Thanks for your work on the timeout, @T045T -- that has helped a lot when a client disconnects / reconnects quickly (such as when refreshing a webpage). Super useful!

For our use case, sometimes a client that's publishing data to the websocket server goes down for more than 10 seconds (i.e., when we're fixing bugs in our code :smile:), after which we restart it. Unsurprisingly, we get the warning error about inbound TCP connections, and then a moderate, random wait until other long-running ROS nodes are able to access to the published data (such as rostopic echo).

I could of course make the timeout longer, or disable unsubscribing all together. I was curious if anyone has also found any other solutions though?

Thanks!

closed 2021-09-18 by github-actions[bot]