RobotWebTools/rosbridge_suite #244: rosbridge websocket server terminates abnormally

guillaumeautran, 2016-11-14:
Looks like the last change to rosbridge might not be complete as the websocket server is terminating abnormally. 

`
Traceback (most recent call last):
  File "rosbridge_server/rosbridge_websocket", line 183, in <module>
    CallService.services_glob = RosbridgeWebSocket.services_glob
NameError: name 'CallService' is not defined
`

The issue was introduced by: https://github.com/RobotWebTools/rosbridge_suite/commit/630e5669d213ebe9eb3331408fb5eb12e382ea66

and is here: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/scripts/rosbridge_websocket.py#L183

T045T, 2016-11-15:
Sorry about that, I should have caught that one.
Fixed in e36d8a2 along with the default values for the topic/service globs.

closed 2016-11-15 by T045T

guillaumeautran, 2016-11-15:
Thanks Nils!