RobotWebTools/rosbridge_suite #249: rosbridge_websocket fails to process incoming messages after the first time

kjeremy, 2017-01-13:
The first run of `node_simple.js` works but if I run it again I see the following output from `rosbridge_websocket.launch` and the twist messages never send again until I restart the launch file.

> [WARN] [WallTime: 1484339590.495914] Could not process inbound connection: [/rosbridge_websocket] is not a publisher of [/cmd_vel]. Topics are [['/rosout', 'rosgraph_msgs/Log']]{'message_definition': '# This expresses velocity in free space broken into its linear and angular parts.\nVector3  linear\nVector3  angular\n\n================================================================================\nMSG: geometry_msgs/Vector3\n# This represents a vector in free space. \n# It is only meant to represent a direction. Therefore, it does not\n# make sense to apply a translation to it (e.g., when applying a \n# generic rigid transformation to a Vector3, tf2 will only apply the\n# rotation). If you want your data to be translatable too, use the\n# geometry_msgs/Point message instead.\n\nfloat64 x\nfloat64 y\nfloat64 z', 'callerid': '/rostopic_6691_1484339001563', 'tcp_nodelay': '0', 'md5sum': '9f195f881246fdfa2798d1d3eebca84a', 'topic': '/cmd_vel', 'type': 'geometry_msgs/Twist'}

This is on indigo, Ubuntu 14.04.

I am listening to the messages with `rostopic echo /cmd_vel`

plieningerweb, 2017-01-19:
Same issue here

robobit, 2017-03-16:
I observe the same behavior. I do get the messages on repeated runs but it can take up to 45 seconds. Once it connects after this long time I also see some "old" messages that seem to have been buffered somewhere.

subvertallchris, 2017-11-21:
With both Jade/Ubuntu 14.04 and Kinetic/16.04, we see this as soon as a JS client abruptly disconnects from the socket. 

Using Jade/14.04, we were working from a fork of this repo for some time now and found that commenting out the following lines was an effective workaround:

https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/publishers.py#L313
https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/publishers.py#L322-L324

We're naturally concerned about what unsafe things might be happening as a result, but our system stays online, so it's hard to argue with those results. I am setting up our Kinetic/16.04 infrastructure now and have not yet tested this workaround, was hoping it had been fixed in the latest release. Going to give it a shot now.

T045T, 2018-01-08:
Have you retried after #247 was merged? That should have some effect

subvertallchris, 2018-01-08:
Have tried, no change, unfortunately.

gbanusi, 2018-03-08:
I had a similar error with tf2:

_[WARN] [1520507044.306192]: Could not process inbound connection: [/arm_fix] is not a publisher of [/tool]. Topics are [['/rosout', 'rosgraph_msgs/Log'], ['/tf', 'tf2_msgs/TFMessage']]{'callerid': '/tool_fix', 'md5sum': '5b83d17121a77a6856144c74d9af68ba', 'message_definition': 'float32 yaw\nfloat32 pitch\nfloat32 roll\n', 'tcp_nodelay': '0', 'topic': '/tool', 'type': 'rfccc/Rotate'}_

sometimes my code would work and sometimes it gives me the above error.
I wanted to send only one message and then unregister the publisher and subscriber.
**For me**, the problem was that I have unregistered the publisher immediately after sending the message to a topic:

```        self.pub = rospy.Publisher("/" + component, msg_type, queue_size=3)
        print(self.id + ' ', end='')
        while self.pub.get_num_connections() == 0:
            print('-', end='')
            rospy.sleep(0.2)
        print('> ' + component)
        self.pub.publish(message)
        self.pub.unregister()
```

but after adding 

```
       while self.pub.get_num_connections() != 0:
            rospy.sleep(0.05)
```

the result code was like:

```
        self.pub = rospy.Publisher("/" + component, msg_type, queue_size=3)
        print(self.id + ' ', end='')
        while self.pub.get_num_connections() == 0:
            print('-', end='')
            rospy.sleep(0.2)
        print('> ' + component)
        self.pub.publish(message)
        while self.pub.get_num_connections() != 0:
            rospy.sleep(0.05)
        self.pub.unregister()
```

Haven't seen the error since...
Have you checked if you unregister your publisher before the message was received?

ablakey, 2018-10-16:
Has there been any progress on this issue? I'm beginning to run into it as well. Unfortunately I cannot guarantee that clients will shut down properly.

NoAxioms, 2018-11-27:
I can still reproduce the error. The commenting out suggested by @subvertallchris works, but doing this will cause an error if you change the type of a publisher. That is, if you have a topic that you change from a Point to a PointStamped, rosbridge will yell at you that the topic already exists with type Point.

subvertallchris, 2018-11-27:
Is it possible that we're disconnecting clients incorrectly? Using [roslibjs](https://github.com/RobotWebTools/roslibjs), I'm just calling `close()` on the connection, which in turn calls `close()` on the websocket. Is there anything else we should should be doing to safely close the connection?

ablakey, 2018-11-27:
I don't believe so. I've tried playing around with what options there are, such as unadvertising topics before exit and nothing changes server behaviour. 

Websocket servers need to be able to gracefully handle abrupt closures of connections. Particularly if you're serving to browsers where you can't guarantee any form of cleanup on application exit. I've got hundreds of users that will just close their phone or laptop and bam, interrupted connection and closed socket after a timeout.

subvertallchris, 2018-11-27:
Of course, I wasn't suggested that doing something other than closing the connection is what it _should_ do, just trying to figure out what could account for some of us seeing this and not others.

ablakey, 2018-11-27:
For sure! I've tried all kinds of hard and gentle cleanups to no luck. I'm going to keep poking at what the delta might be between when it works for me and fails to work at some of our fleet deployments.   Really spams the logs!

jbnunn, 2019-07-02:
Anyone find a solution to this?

Brabalawuka, 2019-07-22:
Is there a better work out fo this issue? Still suffering from the same problem with my android applciaiton implementing rosbridge

subvertallchris, 2019-07-22:
What's weird is that I've been using my fork of the repo for _years_ and I'm not aware of any problems as a result. It makes me wonder whether that code is really necessary and if the solution might just be to remove them entirely?

ablakey, 2019-07-22:
I'm pretty desperate for a fix. I have the above issue, as well as a sibling issue that *might* be related: certain topics can no-longer be published to until I reboot Rosbridge Websocket on the server.

I've got about 50 test robots I can deploy a fix to, then about a thousand I can further deploy to with a fix.  Hopefully this would give us plenty of soak-in to validate any sort of fix.

@subvertallchris if you wouldn't mind pointing to the precise code you think is unnecessary, I can begin getting some test hours on it.

subvertallchris, 2019-07-22:
@ablakey yeah, those are the lines. It's in a private repo so I can't link you to it but here's a screenshot of the diff.

[image]

I made this change on Nov 15, 2017 and have been working without any trouble since then. I also haven't updated my branch since then so it's possible that future changes relying on this behavior would be broken.

ablakey, 2019-07-22:
Thank you, this helps.

I cannot commit to getting this done due to the nature of business factors. But I'm hoping and optimistic to deploy it to the local fleet this week. Will follow-up when I have useful things to share.

subvertallchris, 2019-07-22:
Glad to help, really hope it works out! I would have liked to take the time to dig into it ages ago, I just haven't been able to prioritize it since we've been stable and our needs from this library haven't changed. Happy to help test a patch or review if you make any progress.

mvollrath, 2019-07-22:
If removing those lines fixes the problem, you might try updating rospy, there was an issue fixed for https://github.com/RobotWebTools/rosbridge_suite/issues/165

Removing those lines introduces a memory leak, just so we are all aware.

subvertallchris, 2019-07-22:
It looks like the fix for that was merged back in 2015. I rebuilt my systems a few times and tried in both Kinetic and Melodic, same problem. Could there be something keeping some of us on a very old version of rospy?

mjsobrep, 2020-05-05:
No one else has commented in the last year. I am still having this problem. Is the recommended fix to fork the develop branch and change the lines noted above or is there something else that I missed?

mvollrath, 2020-05-06:
#194 might work too and is a smaller change.

mjsobrep, 2020-05-06:
Thanks @mvollrath. Is this a problem that doesn't cause problems for most people? Is there something that just a few of us are doing to cause it?

mvollrath, 2020-05-06:
Seems like it happens when refreshing the webpage when exactly one browser is subscribed to a topic, causing a rapid unregister -> register cycle.  This suggests there's some problem with the implemented delay before unregistering the rospy Subscriber down in the library.

Never unregistering topics is dirty and I don't want to do it, but it fixes the problem reliably.  Deleting the subscriber might also fix the problem reliably, but I haven't tested it since 2015.  Fixing the unregister timeout logic would be ideal.

mjsobrep, 2020-05-06:
Cool, thanks for the explanation.

transis22, 2020-05-06:
@mvollrath  I've looked at the solution #194 which you mentioned. Please how can I find this ros_comm tcpros_pubsub.py in my Ubuntu system?

mvollrath, 2020-05-06:
You shouldn't need to modify ros_comm, that patch was to fix a memory leak which is related to but separate from this issue.  You would modify
rosbridge_library/src/rosbridge_library/internal/publishers.py and
rosbridge_library/src/rosbridge_library/internal/subscribers.py

But if you're looking for files on your system they will be under /opt/ros/{distro}

transis22, 2020-05-07:
> @ablakey yeah, those are the lines. It's in a private repo so I can't link you to it but here's a screenshot of the diff.
>
> [image]
>
> I made this change on Nov 15, 2017 and have been working without [...]

Okay, thank you. I've modified publishers.py according to this, and still having the problem. Please what do i need to modify in subscribers.py? I'm not clear on that. Sorry, I'm kind of a beginner.

mvollrath, 2020-05-07:
@transis22 here is a full diff https://github.com/EndPointCorp/rosbridge_suite/commit/f3b4791e626a23d6ff2a38111e6783ba3c94ac59

transis22, 2020-05-07:
@mvollrath  Thank you so much for your answers. So I later didn't need to comment out the code. I found out the problem I was having was from overwriting the subscriber in my ros node. I was not supposed to call the subscriber repeatedly. After correcting that, the message publishing works.

gezabohus, 2021-01-20:
Hey everyone,

We are also having this problem, unhacked melodic running on Ubuntu 18.04.

closed 2021-09-18 by github-actions[bot]

acolbourn, 2022-03-21:
@transis22 I was wondering if you could expand on your solution?  We are having the same issue when our android app is closed and connects using roslib again.  It results in weird lags in the commands being sent.