RobotWebTools/rosbridge_suite #256: Glob filter logic not working with last kinetic binaries

doisyg, 2017-01-25:
Hi,
First, thank you for the great work behind rosbridge. 
We would like to use the glob filter logic for security reason (having remote access to all rostopic/param/service can be a bit dangerous). With the current kinetic repo (0.7.16-0xenial-20161026-195828-0700), it is not working, though with the last source, it is.
Can someone push in the repo the last version? It would simplify installation and deployment on our side.
Thanks

jihoonl, 2017-01-25:
https://github.com/RobotWebTools/rosbridge_suite/issues/250
https://github.com/ros/rosdistro/pull/13712

I just did. It should be available with next cycle

doisyg, 2017-02-02:
thanks a lot

closed 2017-02-02 by doisyg