RobotWebTools/rosbridge_suite #258: publishing from rosbridge server to client

waternet, 2017-01-31:
Hello, 

Is it possible to write from the rosbridge server to a connected tcp client? From client to server works as expected, but I am not able to send messages in the other direction. I wasn't able to find any documentation about it, so I'm not quite sure about any supported procedure. 

To my understanding, rosbridge is a bidirectional communication service right?

Thanks,

Sanic, 2017-01-31:
Hi @waternet 
I'm currently using the rosbridge tcp service and am able to communicate in both directions.
For example, my client can subscribe to a topic via rosbridge and will then receive published messages from other ros nodes, for example when using "rostopic pub".
- Are you developing your own client or do you use roslibjs?
- Are you sure that you've successfully subscribed to a topic? Checking the output of "rostopic info" on the ROS side may help.

nullPointer90, 2017-04-28:
Hi @Sanic 
I using ros indigo on ubuntu 14.04, client using C# web socket, my client subscribe to a topic "/listener" via rosbridge but don't receive published message.
I using "rostopic info /listener" , Topic "/listener" is subscribed sucessfully.
Please help me.
Thanks.

Sanic, 2017-04-29:
Hi @nullPointer90 
Is this problem somehow related to the opened issue here? 
If not, feel free to create a separate issue :-)

However, i have never used the websocket connection to rosbridge so far. So i think i personally can't help you on this one. But maybe someone else can.

Atthaporn-Por, 2018-07-08:
Hi @Sanic 
I'm using the rosbridge tcp service and i'm facing the same problem
and i've used "rostopic info " and topic is subscribed successfully.

[image]
and this is log on rosbridge.

klintan, 2018-07-09:
Usually this is related to the wrong bson version, 

`pip uninstall bson`

`pip install pymongo`

And rebuild the package. In ubuntu you should also more easily be able to use 
`rosdep install rosbridge_suite`(or similar) to install it.

Using a Mac there seems to be an issue with Tornado +5.x, see file `websocket_handler.py` in the Rosbridge_server package, the function `send_message` I couldn't get the callbacks to work, if you replace `IOLoop.instance().add_callback(partial(self.write_message, message, binary))` with `self.write_message(message, binary)` it worked for me, will create an issue for this, although I'm running on Python 3 so there might be something else as well.

closed 2021-09-17 by github-actions[bot]