RobotWebTools/rosbridge_suite #259: 2 rosbridge server on the same machine?

doisyg, 2017-02-16:
Hi,
Rosbridge is getting more amazing everyday and we use it now for controlling a robot over the web through secure websocket (wss) with mac authentication and topic glob filtering to ensure a minimum of safety. This is working great.
However we also need to communicate with rosbridge localy without the burden of these security measures. Hence the idea of running simultaneously a second rosbridge server node on a different port, this time with simple websocket (ws), no authentification, and no topic filtering.
Is this possible or there is an incompatibility by design? I tried without investigating too much and i get this error when launching the second server:
`Unable to start server: [Errno 98] Address already in use Retrying in 5s.`

ablakey, 2017-02-16:
Hello! 

I believe you can, but I'm not at a ROS machine to test it.

If you look at the roslaunch for websocket, you'll see that a `port` argument is passed in, or a default is used.  https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/launch/rosbridge_websocket.launch

So try this:

`roslaunch rosbridge_server rosbridge_websocket.launch port:=9091`

doisyg, 2017-02-16:
Once more, I wrote too fast. My second rosbridge server was using a port already in use by another application. Everything seems to work great. I will test later if i notice any conflicting issue in the topic filtering.

closed 2017-02-16 by doisyg