RobotWebTools/rosbridge_suite #262: Missing dependency on bson

VictorLamoine, 2017-02-18:
```
Traceback (most recent call last):
  File "/home/victor/code/catkin_workspace/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket", line 43, in <module>
    from rosbridge_server import RosbridgeWebSocket
  File "/home/victor/code/catkin_workspace/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/__init__.py", line 1, in <module>
    from websocket_handler import RosbridgeWebSocket
  File "/home/victor/code/catkin_workspace/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py", line 42, in <module>
    from rosbridge_library.rosbridge_protocol import RosbridgeProtocol
  File "/home/victor/code/catkin_workspace/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/rosbridge_protocol.py", line 33, in <module>
    from rosbridge_library.protocol import Protocol
  File "/home/victor/code/catkin_workspace/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py", line 35, in <module>
    import bson
ImportError: No module named bson
```

T045T, 2017-02-18:
```python-bson``` is listed in rosbridge_library/package.xml, so it should be installed by rosdep using ```rosdep install rosbridge_library```. I just tried that, and had to do ```rosdep update``` before, to make sure rosdep has a definition for ```python-bson```.

VictorLamoine, 2017-02-18:
I totally agree with you; the current installation instructions are correct.
But this does not mean you don't have to check that `bson` is on the system in CMake.

This is a minor issue.

T045T, 2017-02-18:
I'm not sure that's true - for one, the [catkin documentation](http://docs.ros.org/jade/api/catkin/html/howto/format1/python_module_dependencies.html) doesn't say anything about adding Python dependencies to ```CMakeLists.txt``` - though I guess we should eventually update to [package format 2](http://docs.ros.org/jade/api/catkin/html/howto/format2/python_module_dependencies.html) and ```exec_depend``` instead of ```run_depend```. You'll note the C++ dependency sections *do* mention ```CMakeLists.txt```, so I'm pretty confident we're following the guidelines.
Also, as you mentioned, FindPythonLibs looks like it's CMake 3 only, but virtually every catkin ```CMakeLists.txt``` (including ours) is requires CMake 2.8.3 and up, so we'd be looking at compatibility issues, too.

gavanderhoorn, 2017-02-25:
I agree with @T045T: Python modules should not be `find_package(..)`d. That is only for C/C++.

closed 2017-02-27 by VictorLamoine