RobotWebTools/rosbridge_suite #267: Putting rosbridge_library in pip

cjds, 2017-03-03:
Rosbridge library is designed as a standalone component. Can we pull it into pip, so that we can use just the library's functions, as opposed to using it with the whole suite

rctoris, 2017-03-03:
:+1: would also allow for writing rosbridge implementations outside of catkin if someone wanted.

cjue, 2017-03-03:
Sounds like a good idea. We have written alternative rosbridge handlers that depend on rosbridge_library and rosapi, but not on rosbridge_server, so this would have been useful.

cjds, 2017-03-08:
Okay, so I did some testing. The only things that I can make out that needs changing is a way to set up topic and security globs without the roslaunch doing it, and a requirements.txt needs to be added. I would create a repository, but I think it might make more sense to do it in the Robot Web Tools organization

jihoonl, 2017-03-09:
Could you list up things to do? If necessary, I can prepare repo for pip release.

cjds, 2017-03-09:
1. In rosbridge_suite/rosbridge_library/src/rosbridge_library/capabilities/advertise.py Line 79

       topics_glob = ["*"]

I had to change this from `None` to make it work (This is normally set by roslaunch as far as I can tell). However all the other topics_globs if set to `None` seem to work as if there is no security. 

It would be nice, if there was a way to set these from the `__init__` of the capability, but that would probably be out of the scope of this. These are the 3 params set by roslaunch

         <arg name="topics_glob" default="[*]" />
         <arg name="services_glob" default="[*]" />
         <arg name="params_glob" default="[*]" />

2. In rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py Line 52

        if type(d)==bson.Binary: #This is no longer a subtype in the py-bson library.
        # I changed this to
        if type(d)==bson.BSONCoding:

3. Add bson, rospy, fnmatch etc. as requirements

cjds, 2017-03-13:
Point 2 from above was an unnecessary change, I installed the wrong python bson library. 

This PR https://github.com/RobotWebTools/rosbridge_suite/pull/270
 apparently checks if you have the right one, and warns you.

P.S. Do we have a timeline on this?

sradmard, 2017-09-21:
Hi,
I was wondering whether there is any update on this development or not. This implementation is something that would be very handy for me. 
Thank you.

gonzalocasas, 2019-04-25:
I know it's not the same, but we published a python client to rosbridge protocol called `roslibpy` to pip: https://pypi.org/project/roslibpy/

talregev, 2019-06-21:
@cjds Did you make a pr or repository with rosbridge_library as pip?

closed 2021-09-17 by github-actions[bot]