RobotWebTools/rosbridge_suite #271: Unable to accept incoming connection.  Reason: 'bson_only_mode'

raff44, 2017-03-14:
Hello,

I have been trying to create web control interface for baxter robot, and I tried some example packages. 
After launching "roslaunch rosbridge_server rosbridge_websocket.launch" I am trying to establish connection over port 9090 using webpage, it gives my those errors, Every html script I used returned the same error.

[INFO] [WallTime: 1489421469.154949] Rosbridge WebSocket server started on port 9090
[ERROR] [WallTime: 1489421864.967072] Unable to accept incoming connection. Reason: 'bson_only_mode'
[INFO] [WallTime: 1489421864.967882] Client connected. 0 clients total.

Any ideas how to solve that problem ?

jihoonl, 2017-03-14:
I confirm this error happens.

Hi @Sanic, do you have any idea on this?

jihoonl, 2017-03-14:
How to test
* runs rosbridge_websocket
* open examples/simple.html in roslibjs

Sanic, 2017-03-14:
Hi!
I suspect that this has something to do with the Initialization of the RosbridgeProtocol instance in https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/src/rosbridge_server/websocket_handler.py (and the UDP handler) . In the TCP version, the bson_only_mode Parameter will be passed to the RosbridgeProtocol ctor: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/src/rosbridge_server/tcp_handler.py#L34

I don't have access to my Rosbridge Box right now, but i will check it asap.

Sorry for the inconvience.

ablakey, 2017-03-14:
There's a KeyError on https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/protocol.py#L105

I'm raising a pull request with a trivial fix. But it may be that the true fix exists further upstream from that line of code.  Ie. prevent `parameters['bson_only_mode']` from ever being undefined.

ablakey, 2017-03-14:
Also note that this error gets swallowed by a catch-all: https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_server/src/rosbridge_server/websocket_handler.py#L72

Probably worth considering reporting a more descriptive error such as including the error type. ie. `Unable to accept incoming connection. Reason:  KeyError: bson_only_mode`.  Or better yet, don't catch all exceptions and allow the node to crash when a software bug like this is introduced.

jihoonl, 2017-03-14:
I just merged the quick fix from @ablakey. devel branch should work as of now. Yeah I agree. It would be more easy to debug if exception catches only known issues and let others crash.

jihoonl, 2017-03-15:
Issue resolved by #273

closed 2017-03-15 by jihoonl