RobotWebTools/rosbridge_suite #278: Support ROS2.0 in rosbridge

minggangw, 2017-04-07:
Hello,

As the ROS2.0 Beta1 has been released at the end of last year, and we'll have beta2 in the middle of this year, finally 1.0 Version in Dec ([ROS2.0 Roadmap](https://github.com/ros2/ros2/wiki/Roadmap)). I think it is the right timing to begin to support ROS2.0 in rosbridge.

Actually I have done some work on enabling ROS2.0 on rosbridge, I propose two solutions:

- Using `rclpy` as backend, just like what we did now.
- Implementing a nodejs client for ROS2.0, which we can leverage to connect to ROS and runs under Node.js env.

What do you think? suggestions?

Thanks in advance!

kenny-y, 2017-04-10:
I've heard that this project is recruiting maintainers... I'd like to become one and contribute code for ROS 2.0 bridge.

BTW. minggangw is my colleague.

jihoonl, 2017-04-10:
both sounds very nice to me. option 1 will allow accessing a robot directly from browser just as before. Whereas option 2 will give full system access as a native client of ROS2.0.

I just create a [new repo](https://github.com/RobotWebTools/ros2bridge_suite-experiment) and added both of you as outside collaborator. Feel free to implement the ros2.0 version of rosbridge.  And Please let me know if anything is needed.

Nodejs client would be nicer if it can go as official ROS2.0 support. I would try that first.

kenny-y, 2017-04-10:
Hi Jihoon,
Thanks very much for your response, I’m very happy that I can have this opportunity to contribute to RTW/ROS community.
We (Minggang  & me) are from Intel Software & Service Group, and we’re fulltime developers working in Open Source Technology Center (OTC). Currently we’re finding opportunities in the combination of web technology + robotics technology, which makes RWT fall right into this category.
I noticed that you mentioned some design consideration… do you have a preferred way to discuss it? It might take us a few days to work out some draft design/etc. before make it happen…
BTW. We’re in UTC + 8:00.
-Kenny

jihoonl, 2017-04-10:
Hi @kenny-y, I would prefer to have a discussion in github issue so that others also can participate. Feel free to take enough time. We are not in hurry :) I'm in CET(UTC+1 +1) timezone.

jihoonl, 2017-04-10:
Also if you will have a draft document, feel free to use the new repo to share `rst` files.

kenny-y, 2017-04-10:
Hi @jihoonl, great. I also felt the same way :)

+ @minggangw So let's get our hands dirty...

minggangw, 2017-04-10:
Hi Jihoon, thanks for your quick response. As Kenny said above, we, who are focusing on web technology, are expecting to push web into the robotics. We find RWT is a perfect combination and start point.

I think we can reach consensus at the very early stage if possible, which will guarantee we are on the right direction.  Thanks!

@kenny-y let's make the 2.0 era start :)

nuclearsandwich, 2017-04-10:
> Nodejs client would be nicer if it can go as official ROS2.0 support. I would try that first.

Yeah, I think building nodejs bindings from [ros2/rcl] after the same fashion that the [python] and [C++] bindings are done would be a better way to incorporate nodejs support.

[ros2/rcl]: https://github.com/ros2/rcl
[python]: https://github.com/ros2/rclpy
[C++]: https://github.com/ros2/rclcpp

minggangw, 2017-04-11:
>bindings are done would be a better way to incorporate nodejs support.
That is what we want exactly! If we can incorporate nodejs, then we can leverage Electron/NW.js to develop web based ROS2 apps rapidly, many thanks!

kenny-y, 2017-04-13:
I just made a quick experiment work -- I wrote a Node.js app that constantly publishes a string message (as a substitute to [C++ minimal publisher](https://github.com/ros2/examples/tree/master/rclcpp/minimal_publisher)), and then used [C++ subscriber](https://github.com/ros2/examples/tree/master/rclcpp/minimal_subscriber) to listen and log the string message.

To make the Node.js app work, I wrote a Node.js C++ add-on, providing some non built-in API to Node.js environment by myself. I used `node-gyp` to build the add-on that depends on `rcl` and other necessary support library for message. So the call graph is:

`JS app --> Node.js C++ add-on --> rcl`

I guess this is a practical way to quickly build a Node.js client, and then write the bridge on top of it. The Node.js client itself is useful for several other scenarios...

Note that the above can be perceived as a proof of concept for making design decision.

kenny-y, 2017-04-14:
A diagram to illustrate the role of Node.js C++ Add-on

[image]

nuclearsandwich, 2017-04-14:
Should we create a node-ros2 / rosclnode repository for this work?

kenny-y, 2017-04-14:
That works for me... a separate repo is better than putting client + bridge into one, I guess.

BTW. to avoid misunderstanding, we might need to use `nodejs` in repo name, to suggest it's different to a ROS `node`

minggangw, 2017-04-14:
We have the first step now. @nuclearsandwich I think `rclnodejs' is fine, which means that it's based on rcl and written in JavaScript with nodejs. This also follows other clients' convention, like rclcpp/rclpy/rclcs(c# client).

kenny-y, 2017-04-17:
Just did another quick experiment, now a Node.js subscriber prototype is working. I verified the following combinations

```
Node.js publisher --> Node.js subscriber
Node.js publisher --> C++ subscriber
C++ publisher --> Node.js subscriber
```

My next target is `service`.

P.S. my plan is to first make prototypes working for all features, and then do the API design, based on what we've learned from the prototypes.

minggangw, 2017-05-04:
@kenny-y and I have been working on the prototype of rclnodejs in these two weeks, and we have done most important functions. Below, I show the detailed status of our work:

Features | Dev Status | Unit Test
--------- | :----------: | :--------: |
node | done | passed
publisher | done | passed
subscriber | done | passed
client | done | passed
service | done | passed
timer | done | passed
wait | done | passed
time | not started | not started
qos | not started | not started

Since we are proceeding with our internal open source assessment, we have to wait it done and have the permission to push the code. We are planning to implement the `rclnodejs` in the first step, then integrate it with the ros2bridge, which will also adapt to Node.js. 

I think we'd better have a separate repo for `rclnodejs`, as it's a standalone ROS2.0 client and ros2bridge will depend on it. Should it be located under RobotWebTools or ROS2 org?

nuclearsandwich, 2017-05-04:
> I think we'd better have a separate repo for rclnodejs, as it's a standalone ROS2.0 client and ros2bridge will depend on it. Should it be located under RobotWebTools or ROS2 org? @nuclearsandwich

I can't speak for @jihoonl and the veteran RWT team members, but I think either RobotWebTools or a new organization is the best place for the repository as even though I'll be helping out where I can, rclnodejs will still be a community supported ROS 2 client library similar to ros2_java or rclcs.

jihoonl, 2017-05-11:
Hey @tfoote, here @kenny-y  and @minggangw are developing `rclnodejs`, the nodejs version of  ros2 client library. Can they have a repo in [ros2 organization](https://github.com/ros2) to implement `rclnodejs` as a pure ros2 client?

tfoote, 2017-05-16:
@jihoonl That's great to hear. As with ROS1 we don't want to require centralizing development and want to encourage a community sense of ownership by not requiring placement of the repos in a central location. For example there's already [rclcs](https://github.com/firesurfer/rclcs) in firesurfer's account. I'd encourage you to set it up whereever is most convenient for development now and in the future we can consider moving it.  @kenny-y @minggangw I'd also encourage you to start a thread on discourse to increase visibility and possibly get contributors. https://discourse.ros.org/c/ng-ros

minggangw, 2017-05-18:
Thanks for your suggestions @tfoote , we will hold the code under the RWT umbrella firstly, and hope to improve the influence of Node.js in ROS2.0 ecosystem later.

minggangw, 2017-05-19:
Hi @jihoonl our internal assessment for open source project has been done, so we can expose our code now. I created a folder named [rclnodejs](https://github.com/RobotWebTools/ros2bridge_suite-experiment/tree/master/rclnodejs), where originally I want to put the node.js client `rclnodejs`. Through the negotiation with you, @nuclearsandwich and @tfoote, I prefer to have it locate in a separate repo, and the [ros2bridge_suite-experiment](https://github.com/RobotWebTools/ros2bridge_suite-experiment) will utilize the `rclnodejs` as a dependency, which make it more reasonable instead of integrating them together. So would you help me to create the repo under the RobotWebTools organization and add @kenny-y & me as collaborators? Thanks!

jihoonl, 2017-05-19:
Hello @minggangw, I have prepared a new repo for you. Since you also have an admin access on repo, feel free to play with it. :wink:

closed 2017-06-28 by jihoonl