RobotWebTools/rosbridge_suite #280: subscribe function's queue_length default value is inconsistent between document and code

ledmonster, 2017-05-15:
subscribe function's queue_length default value is inconsistent between document and code.

[Document](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) saids:

> queue_length – the size of the queue to buffer messages. Messages are buffered as a result of the throttle_rate. Defaults to 1.

But [code](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/capabilities/subscribe.py#L235)'s default is 0.

```
        subscribe_args = {
          "sid": sid,
          "msg_type": msg.get("type", None),
          "throttle_rate": msg.get("throttle_rate", 0),
          "fragment_size": msg.get("fragment_size", None),
          "queue_length": msg.get("queue_length", 0),
          "compression": msg.get("compression", "none")
        }
```

Which one should be standard ?

mvollrath, 2017-09-15:
Also defaults to zero in [roslibjs](https://github.com/RobotWebTools/roslibjs/blob/45792e1d90ac2c201513dc9f1f508d24c7da6b0c/src/core/Topic.js#L37)

pedro1713, 2017-10-10:
Queue size = 0  would indicate an infinite queue. I would assume that defaulting to an infinite queue is reasonable to avoid missing messages, though not entirely sure this is the original intent.
Maybe we should correct the documentation rather than the code.

PR #302 (2017-11-20): Fix subscrider's default queue_length in documentation to match code
Fix #280 by modifying the document to match code. Subscriber's default queue_length is 0.
commit f0d35979 (2017-11-20): Fix subscrider's default queue_length in documentation to match code. Fixes #280 (#302)
closed 2017-11-20 by jihoonl