RobotWebTools/rosbridge_suite #294: Subscribing to /tf only delivers delayed messages

Maverobot, 2017-09-06:
Hello,

as the title suggests, I am experiencing a weird problem of delayed /tf messages. The delay was between 5 seconds to 20 seconds. /odom topic is also delayed once the /tf topic is subscribed. Both issues happened only on the rosbridge client side, meaning using "rostopic echo" still gives a proper /tf and /odom.

Did anyone have similar problems?

jihoonl, 2017-09-06:
It is because your socket is not fast enough to consume the incoming data. Try (tf2_web_republisher)[https://github.com/RobotWebTools/tf2_web_republisher] for tf streaming.

Maverobot, 2017-09-09:
Thanks. [tf2_web_republisher](https://github.com/RobotWebTools/tf2_web_republisher) works like a charm.

closed 2017-09-09 by Maverobot