RobotWebTools/rosbridge_suite #298: Rosbridge could not process inbound connection

ThotAlion, 2017-10-11:
Hello,

I have an issue : 
- I launch a topic named /flow
- I launch rosbridge
- I open a web page which subscribe and publish on this topic (and all is OK)
- I close this page and 1 second after, I reopen it, and all is OK
- I close this page and after a while (5 seconds), in console I have : 
[WARN] [1507736546.991884]: Could not process inbound connection: [/rosbridge_websocket] is not a publisher of [/flow]. Topics are [['/rosout', 'rosgraph_msgs/Log']]
- rosbridge is not publisher on this topic
- I open a web page but it cannot publish on this topic until I refresh several times.

Is it possible to constantly subscribe to the /flow topic to avoid web control loss ?

jbcpollak, 2017-11-16:
I seem to have the same problem

jihoonl, 2017-11-20:
@ThotAlion  This is not a problem in rosbridge but in rospy. Check #138 to avoid the subscription issue.
@jbcpollak I don't think it is the same issue. Can you create an issue with details?

closed 2017-11-20 by jihoonl

chedanix, 2021-05-17:
It's been 3 years since anyone posted here. Here is what I have found today.

1. Rospy still has issues with unregistering, so the root cause is still not fixed. Someone wrote a test for it and you can see it for yourself [https://gist.github.com/ejalaa12/3f0e76d11840549f1463c84f583b7642](url) 
2. The first workaround is the popular "quick hack" that comments out the 3 lines of code in  rosbridge_library/src/rosbridge_library/internal/publishers.py
3. The second workaround is setting the unregister_timeout parameter in the launch file to a huge number.

acolbourn, 2022-03-23:
Thanks chedanix! Only been 1 day so far but setting unregister_timeout to 99999999 has solved it for me.  This is a much simpler solution than commenting out the unregister code in the typical solution bc it can be done from the launch file and it has the same effect.  For reference, the typical solution is to comment like this: https://github.com/EndPointCorp/rosbridge_suite/commit/f3b4791e626a23d6ff2a38111e6783ba3c94ac59

 I also noticed that if I tried to unregister my publisher in my code like `publisher.unadvertise()` , for example, that this triggered the warning.  So I removed .unadvertise() everywhere, added the huge unregister_timeout, and so far so good.  I'll report back if the error occurs again.