RobotWebTools/rosbridge_suite #301: Versioning question

subvertallchris, 2017-11-14:
Hey guys, thanks for all your work on this project. Could someone clarify the versioning scheme? We've been using 0.7.16 and are looking to upgrade. I see that you released 0.8.0 in August. Can we treat this as a major release and assume that there might be breaking changes, while your minor releases (others within 0.8.x) only contain backwards compatible fixes or improvements? Or should each release be treated as potentially breaking? Thank you!

jihoonl, 2017-11-20:
I try to follow semantic versioning. API breaking change or updates in dependency(changes in package.xml) tree will increase minor version and other changes will increase patch version. So, if patch version increases, you can assume that it doesn't break the API.

closed 2017-11-20 by jihoonl

subvertallchris, 2017-11-20:
Understood, thank you.