RobotWebTools/rosbridge_suite #303: Unable to start multiple bridges

RoelHelsen86, 2017-11-16:
We are using rosbridge for remote access to our ros programs and it works great, however we have one problem and that is that you cannot start multiple bridges when we use the unmodified source code.

The problem is that every bridge wants to start a "rosapi" node with the same name.

To solve this, I added an extra argument to the launch files for the different bridges to enable/disable starting a rosapi node:

#add to launch arguments:
  <arg name="first_bridge" default="true" />

#change the node declaration for rosapi as follows:
  <node name="rosapi" pkg="rosapi" type="rosapi_node" if="$(arg first_bridge)">

with this additions, the launch files will launch a rosapi node by default, but it can be disabled by inluding a first_bridge=false argument. ( Iassume that only one rosapi node is necessary)
Ultimately the launch file should detect if there is already a rosapi node and not start one if not neeeded, but I don't know how to do that in the current framework.

RoelHelsen86, 2017-11-20:
I don't have any real reason other than convenience to stay with the default launch files, and indeed creating my own launch files would solve the problem as well.
I am correct that I only need one rosapi node for multiple bridges, right?

I raised the issue because I figured that I would not be the only one that wants to start multiple bridges and it took me some time to figure out what was going wrong by doing so.

jihoonl, 2017-11-20:
> I don't have any real reason other than convenience to stay with the default launch files, and indeed creating my own launch files would solve the problem as well.
I am correct that I only need one rosapi node for multiple bridges, right?

yeah one rosapi should be sufficient for multiple rosbridges unless you are running hundreds.

> I raised the issue because I figured that I would not be the only one that wants to start multiple bridges and it took me some time to figure out what was going wrong by doing so.

I see thanks.  I haven't seen the use case yet. if more people complains about this, i will try to add another launcher(or argument) to run multiple instance of rosbridge.

closed 2017-11-20 by jihoonl

T045T, 2017-11-20:
I'm wondering why only ```rosapi``` caused problems - if you started the launch file multiple times, then the bridge itself would have a duplicate name too.

RoelHelsen86, 2017-11-20:
I did not make multiple bridges of the same type. I needed both a websocket- and a tcp-bridge, otherwise that would have been the case. That's why I only made an extra argument to disable the start-up of the rosapi node.

If I wanted to start multiple bridges of the same type (say 2 TCP bridges on different ports), I just have to give the nodes a different name, right?

T045T, 2017-11-20:
That's correct. As Jihoon mentioned, this is an unusual use case, so please do submit any issues you run into :)