RobotWebTools/rosbridge_suite #304: can't locate node in package

subvertallchris, 2017-11-20:
Hey guys, running into a problem getting up and running. We've been using a slightly older version for some months now by running this repo as a package in our `src` directory. Everything has been working happily but we're moving into production and giving the latest release a shot. I'm running into an error when I run my launch file.

```
ERROR: cannot launch node of type [rosbridge_server/rosbridge_websocket]: can't locate node [rosbridge_websocket] in package [rosbridge_server]
ERROR: cannot launch node of type [rosapi/rosapi_node]: can't locate node [rosapi_node] in package [rosapi]
```

When I follow the "Running Rosbridge" instructions [here](http://wiki.ros.org/rosbridge_suite/Tutorials/RunningRosbridge), it starts normally. My package's launch file looks nearly identical and has worked for months, we just add args for certfile and keyfile. I'm not at all convinced that this is truly a rosbridge_suite bug but I've been unable to find a solution elsewhere so thought I'd go right to the source.

Appreciate your help with this. Thank you!

jihoonl, 2017-11-20:
I just doubled checked that everything works with the latest version. Make sure your catkin workspace is properly set. Are the other packages working?

subvertallchris, 2017-11-20:
Thanks for the quick reply! As far as I can tell, yes. Our custom packages start and I'm seeing no errors with other dependencies, but who knows what's lurking. ;-)

Any advice for checking the workspace? My gut says it's something about a path.

T045T, 2017-11-20:
A few things you can try:
* Run ```roswtf``` on your launch file. 
* ```roscd rosbridge_server``` should take you to where ROS think the package is (this could show you if it's using a system version from ```/opt/ros/...``` or if it doesn't know about the package at all).
* Last but not least, make sure you've ```source```'d your ```setup.bash``` file since the last build. Your environment variables may not be up do date.

Hope one of those helps :)

subvertallchris, 2017-11-20:
Thanks, guys. I just finished rebuilding my workspace and never everything's working. Of course. I'll definitely bookmark those troubleshooting tips for later.

closed 2017-11-20 by subvertallchris

amadojoao, 2019-10-02:
How did you rebuild the workspace?