RobotWebTools/rosbridge_suite #308: rosbridge_tcp send less than 29200Bytes when subscribe /map

wbang8933, 2017-12-27:
I run rosbridge_tcp on my ubuntu14.04 device，and run a python client to subscribe /map data on windows，my map‘s resolution is 1147 X 508，I can only receive 29200Bytes， why？

The following is my python client code：
`import time                              
import socket                            
import simplejson 
import threading

def subscribeMap():
	sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
	#sock.settimeout(10)
	sock.connect(('192.168.1.111', 9090))
	#sock.setblocking(0)

	data = {
		'op':'subscribe',
		'topic':'/map',
		'type':'nav_msgs/OccupancyGrid'
	}

	st = simplejson.dumps(data)
	sock.send(str.encode(st))
	while 1:
		try:
			rtn = sock.recv(65536)
			#print("===>map data: ", rtn)
			print("$$$$$$ map size: %d" % len(rtn),total_size, type(rtn))
		except:
			time.sleep(0.01)
			continue
	sock.close()

if __name__ == '__main__':

    subscribeMap()
    print("app exit")`

T045T, 2018-01-02:
Your code looks like it's missing something (you never set total_size), and I don't know what sort of OccupancyGrid you're working with - can you share some output from your script?

I'd be particularly interested in the actual data you receive - do you get complete JSON objects that are missing data, or [fragments](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#311-fragmentation--fragment--experimental) (you'll have to assemble these)?

wbang8933, 2018-01-05:
@T045T Thanks for you reply，I solve it by change the self.request.send to self.request.sendall in the send_message in tcp_handler.py。
By the way ,you said total_size, how can I set it?

T045T, 2018-01-05:
Ah, that explains it, good catch!

Would you mind making a small PR with that change, so you can get credit for the fix?

```total_size``` is an undefined variable in your example client code, nothing to do with rosbridge itself

T045T, 2018-01-08:
Fixed by #310, thanks!

closed 2018-01-08 by T045T