RobotWebTools/rosbridge_suite #321: bson exception block rosbridge sending out json to websocket

yuan8421, 2018-02-05:
To install rosbridge_suit, I run
`sudo apt-get install ros-indigo-rosbridge-server
sudo pip install bson twisted
`
bson version is **0.5.1**
After subscribe ros topic via web api, the rosbridge is blocked when it tried to send out the json file.
In these two files:

> /ros/lib/python2.7/dist-packages/rosbridge_library/internal/message_conversion.py
> /ros/lib/python2.7/dist-packages/rosbridge_library/protocol.py

It import bson and call bson.Binary:
`
def get_encoder():
    global binary_encoder
    if binary_encoder is None:
        binary_encoder_type = rospy.get_param('~binary_encoder', 'b64')
        if binary_encoder_type == 'b64':
            binary_encoder = standard_b64encode
        elif binary_encoder_type == 'bson':
            binary_encoder = bson.Binary   
        else:
            print "Unknown encoder type '%s'"%binary_encoder_type
            exit(0)
    return binary_encoder
`
And
`
def has_binary(d):
    if type(d)==bson.Binary:
        return True
    if type(d)==dict:
        for k,v in d.iteritems():
            if has_binary(v):
                return True
    return False
`
Here is the problem that python argue that

> 'module' object has no attribute 'Binary'
I did some fix by myself. But it will be good have a official fix for this problem.

Behery, 2018-02-06:
This has been fixed by [this PR](https://github.com/RobotWebTools/rosbridge_suite/pull/315) but it's not in the latest release yet.

yuan8421, 2018-02-06:
Oh, i didn't find this. Good to know. Thanks @Behery!
Close thread.

yuan8421, 2018-02-06:
Close.

yuan8421, 2018-02-06:
close

closed 2018-02-06 by yuan8421