RobotWebTools/rosbridge_suite #333: Release date for ROS Melodic

nlynn12, 2018-05-25:
What’s the release date for the rosbrigde server on Ubuntu 18.04 (ROS Melodic)?

cberner, 2018-06-10:
Just wanted to add that I tried running all the tests on 18.04 and they passed

ramezanifar, 2018-06-29:
Can you please release the rosbridge_suit for ROS Melodic?

jihoonl, 2018-07-01:
[rosauth](https://github.com/GT-RAIL/rosauth) is not being released in order to release rosbridge in melodic.

@rctoris Can you release rosauth?

flynneva, 2018-07-08:
....waiting on this too.  Any ideas when this will be released?

pjreed, 2018-07-11:
rosauth has been released (https://github.com/ros/rosdistro/pull/18413), so this should be clear to go now.

closed 2018-07-18 by jihoonl

jihoonl, 2018-07-23:
https://github.com/ros/rosdistro/pull/18488

ramezanifar, 2018-07-26:
> ros/rosdistro#18488 <https://github.com/ros/rosdistro/pull/18488>

Hi
In Ubuntu 18.04

sudo apt-get install ros-melodic-rosbridge-server

results in

E: Unable to locate package ros-melodic-rosbridge-server

Does that mean something was not completed in the release?
Thank you in advance

jihoonl, 2018-07-30:
http://repositories.ros.org/status_page/melodic_default.html?q=rosbridge

It should have been released. Did you apt-get update?

ramezanifar, 2018-07-31:
Hello Jihoon
My problem is fixed.
Thank you for your help
Regards