RobotWebTools/rosbridge_suite #336: Nothing returned from rosbridge webserver

rachelspykerman, 2018-06-19:
Hi,

I am running rosbridge on a raspberry pi 3 model B+ (Stretch), and using roslibjs to create a simple webpage that can subscribe to a topic. I am unable to retrieve anything from the rosbridge server though. I know my code is correct because I had this previously set up on a raspberry pi model B (not B+) which ran Jessie (not Stretch), and everything worked fine. I was able to get topics from the server and subscribe to topics. I recently moved over to the 3 B+, installed rosbridge from source and a few extra packages, and now I'm no longer able to retrieve a list of topics or receive subscribed data.

I've done a lot of reading on forums that ran into the same issue (like this one: https://github.com/RobotWebTools/rosbridge_suite/issues/198) which suggested to install pymongo and not bson. So I uninstalled bson, and did a "sudo pip install pymongo", however, this did not resolve my issues.

My install steps are outlined below:
	cd ~/catkin_ws/src
	git clone https://github.com/RobotWebTools/rosbridge_suite.git
	git clone https://github.com/GT-RAIL/rosauth.git
	cd ~/catkin_ws
	catkin_make

When running "roslaunch rosbridge_server rosbridge_websocket.launch", I ran into missing packages: tornado, bson, and python-twisted. I ran the following to install these packages:
        sudo pip install tornado
        sudo pip install pymongo
        sudo apt-get install python-twisted

The rosbridge server now starts without any issues, I can connect to port 9090, but I do not receive any data back from the server. Example, the following code use to return a list of topics (when running on Jessie and older pi model):
`    function getTopics() {`
`      var topicsClient = new ROSLIB.Service({`
`        ros : ros,`
`        name : '/rosapi/topics',`
`        serviceType : 'rosapi/Topics'`
`      });`
`      var request = new ROSLIB.ServiceRequest();`
`      topicsClient.callService(request, function(result) {`
`        console.log("Getting topics...");`
`        var resultsSorted = result.topics.sort();`
`        for(var i=0; i < resultsSorted.length; i++) {`
`            //console.log(resultsSorted[i]);`
`            var opt = document.createElement("option");`
`            document.getElementById("topicNameSub").innerHTML += '<option id="' + i + '">' + resultsSorted[i] + '</option>';`
`        }`
`      });`
`    };`

The following are the versions of some of the packages I installed:

Raspberry pi: model 3 B+ running Raspbian Stretch version 9.4
python: 2.7.13
pymongo: 3.6.1
tornado: 5.0.2
twisted: 16, 6, 0

How can I fix this issue? I don't get any errors messages in the rosbridge log files. It seems to just silently ignore any requests to the server. Please let me know if you need any other information. I really appreciate your help!

Rachel

Behery, 2018-06-19:
Maybe you aren't exposing any topics? there are parameters that control which topics/services/parameters are exposed in the `rosapi`:
```
<node name="rosapi" pkg="rosapi" type="rosapi_node">
        <param name="topics_glob" value="$(arg topics_glob)"/>
        <param name="services_glob" value="$(arg services_glob)"/>
        <param name="params_glob" value="$(arg params_glob)"/>
</node>
```

rachelspykerman, 2018-06-19:
So when I run "roslaunch rosbridge_server rosbridge_websocket.launch", it tells me the following:

SUMMARY
========

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

ROS_MASTER_URI=http://192.168.8.3:11311

process[rosbridge_websocket-1]: started with pid [3777]
process[rosapi-2]: started with pid [3778]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [1529432471.034303]: Rosbridge WebSocket server started on port 9090

So it seems from the parameters list that /rosapi/params_glob, /rosapi/services_glob, and /rosapi/topics_glob are all being exposed (since they are set to *). Maybe I'm misunderstanding that though. If it's being exposed, how do I go about setting it up?

Behery, 2018-06-19:
you're right. it means everything should be exposed. To see where the problem is, can you try calling the rosapi services themselves? 

```rosservice call /rosapi/topics
    rosservice call /rosapi/services
```
it should give you the same output as `rostopic list`

TechnoX, 2018-06-19:
Seems like I have the same problem. The only solutions are referring to the bson bug, but as far as I can see I have the correct version (from pymongo). 

Tried your code above and no topics were reported back. 

On my Ubuntu where I run roscore and rosbridge: 
pymongo (3.6.1)
Python 2.7.12
tornado (5.0.2)
Twisted (18.4.0)

Linux 4.13.0-45-generic #50~16.04.1-Ubuntu SMP Wed May 30 11:18:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

rachelspykerman, 2018-06-19:
I finally got it working! So I uninstalled pymongo, tornado, and python-twisted (and any other packages I had manually installed that had caused the roslaunch not to run). After uninstalling, I ran rosdep install, which will install all dependent packages needed for your current ros setup:
         rosdep install --from-paths src --ignore-src -r -y

After doing this, my problems were resolved :) I'm now able to retrieve topics and subscribe to topics.

TechnoX, 2018-06-20:
Thanks! That resolved the issue for me as well! 
Strange that we happened to get the same error at the same time! I have no idea what was the cause, probably some inconsistencies between packages.

closed 2018-06-20 by Behery

mpena2099, 2018-08-19:
Thank you very much, @rachelspykerman.

ericq, 2019-09-23:
tornado version must be in 4.5.3.
pip uninstall tornado
pip install tornado==4.5.3
// this is so bad now.. because my jupyter notebook won't work with such an old version of tornado.

mvollrath, 2019-09-23:
If you use rosbridge_server from the develop branch it will run Autobahn instead of Tornado.  Should be released for kinetic+ soon.