RobotWebTools/rosbridge_suite #344: Can WebSocketSharp to Coo

p-buddy, 2018-07-24:


asimonov-hitachi, 2019-01-11:
hi! I am seeing the same issue - using Unity based LG Silicon Valley Lab simulator sending data to ros via rosbridge. Some images are published at 15Hz, but only arrive at a ROS topic (running in container on same host) at 5Hz.

Anyone has any ideas how to speed this up?

solernou, 2019-02-28:
I am having a similar problem, that I posted in ROS#'s site, see:
  https://github.com/siemens/ros-sharp/issues/161

Any help would be very welcome.

mvollrath, 2019-04-04:
I recommend the new CBOR compression type for binary data such as OccupancyGrid, it will be orders of magnitude faster.  You will need to implement decoding in C# yourself, though.  The OccupancyGrid data will come through as a tagged bytestring, tag 72.  See [CBOR typed array tags](https://datatracker.ietf.org/doc/draft-ietf-cbor-array-tags/?include_text=1)

Caveats, 2020-04-30:
Hey - did you resolve this? Looking at a similar function for Unity.

closed 2021-09-17 by github-actions[bot]