RobotWebTools/rosbridge_suite #345: Port robot_web_tools to adapt to ROS 2.0

minggangw, 2018-07-27:
Dear all,

The second ROS 2.0 stable release, [Bouncy](https://github.com/ros2/ros2/wiki/Release-Bouncy-Bolson), was published recently, meanwhile I noticed the roadmap of next release, [Crystal Clemmys](https://github.com/ros2/ros2/wiki/Roadmap), is available also. I want to point out one item of the whole roadmap:

>Port robot_web_tools

I created this thread to collect ideas about this action, and I asked some questions on [ROS 2 repo](https://github.com/ros2/ros2/issues/529) and got some explanations from @dirk-thomas already. If you have some plans or ideas, please discuss it below.

Thanks!

dirk-thomas, 2018-08-13:
We (OSRF) would like to contribute ROS 2 ports of some of the packages in Robot Web Tools - not only in this repository but also in other ones under this org unit. Commonly we prefer to create a `ros2` branch in the upstream repository in order to keep the code as close as possible to the ROS 1 code.

Since neither of us is a maintainer of these repositories we wanted to check with you if that is desired and "ok" or if you prefer us to use forked repositories instead.

jihoonl, 2018-08-14:
Hi @dirk-thomas, it is no problem giving one of you a maintainership and create a ‘ros2’ branch as long as someone maintains it.

Also, since @minggangw and his team are actively working on ros2 version of rwt, I would like to follow his team’s opinion.

minggangw, 2018-08-14:
Hi @dirk-thomas and @jihoonl very glad to hear that the ROS community is moving forward to implement the web components for ROS 2.0. It's no problem to create a `ros2` branch from my point.

Beyond that, we already have rclnodejs and ros2-web-bridge for ROS 2, and you can consider them to see if they are useful for OSRF's work. As @jihoonl said, we are working on them actively and willing to contribute to either the current repo or the branch to be created by you.

dirk-thomas, 2018-08-14:
> it is no problem giving one of you a maintainership and create a ‘ros2’ branch as long as someone maintains it.

We would certainly do the initial porting and making sure the packages work, continue to work and will be available in ROS 2 distributions. I can't say anything about feature development in the future. That usually depends on the direction of projects. But the work should help to makes the existing functionality available in ROS 2.

> we already have rclnodejs and ros2-web-bridge for ROS 2, and you can consider them to see if they are useful for OSRF's work

We have looked at both repositories. Since we need additional functionality which is already provided by other packages in the Robot Web Tools org unit we decided to convert these to ROS 2 (which should be fairly minimal effort).

jihoonl, 2018-08-16:
@dirk-thomas I have sent you the team invitation. You should be able to create a branch for most of repos in the org. Feel free to create branches as you like.

In the meantime, I'm in the discussion about how to move forward as my time is getting limited for maintenance. Once I have a conclusion, I will let you know.

dirk-thomas, 2018-08-16:
> I have sent you the team invitation.

Great, thanks!

> Feel free to create branches as you like.

I created a `ros2` branch - for now it is at the same commit as the `develop` branch.

dirk-thomas, 2018-08-20:
These are the current PRs (in progress):
* RobotWebTools/rosbridge_suite#351
* GT-RAIL/rosauth#18
* RobotWebTools/web_video_server#69
* GT-RAIL/async_web_server_cpp#12

minggangw, 2018-08-21:
@dirk-thomas I don't have specific things to be added here and we can track the porting issue in the separate repositories. Please go ahead to close this issue, thanks!

closed 2018-08-21 by dirk-thomas

jubeira, 2019-02-01:
Just as a heads up, we are currently working on this on this fork: https://github.com/ros2/rosbridge_suite. The idea is to keep the work in progress in the fork and once it takes shape move it to `ros2` branch here as Dirk mentioned before.

Here's a [meta ticket](https://github.com/ros2/rosbridge_suite/issues/1) to track the overall progress in case anyone's interested.

dirk-thomas, 2019-02-01:
@jubeira Please target the branch of #351 so that there is only a single PR to be merged into `ros2` at the end.

jtbandes, 2021-07-15:
Hi, what's the status of this project? I see the `ros2` branch has had some work done, but it doesn't look like rosbridge-suite has been part of any ROS 2 releases yet.

jubeira, 2021-07-16:
Hi @jtbandes,
If I remember correctly I could do some basic back and forth pubsub between ROS2 and roslibjs with the work on that branch, but it needed some more work to be release-ready.
I'm no longer involved in that track, but perhaps that branch is a good starting point to take over if you need that functionality.

jtbandes, 2021-07-16:
Actually I found that the package _was_ released in Dashing, but hasn’t been in recent ROS 2 releases…do you know why?