RobotWebTools/rosbridge_suite #352: "Message too big" - Error

PatrickNa, 2018-08-20:
Dear Community,

I need to establish a rosbridge websocket-connection between the machine running ROS and an Android tablet. The solution of mine worked well with rosbridge installed on Ubuntu 16.04 and Kinetic. However on Ubuntu 18.04 and Melodic I receive a "message too big" - error.

A little more insights: I am able to connect to the rosbridge server with the tablet, but subscribing or publishing to a topic throws the aforementioned error and the client gets disconnected. The version I use refers to commit `e72b736f3a218883367dc2ec0bc3d05a8b581b48` (latest in _develop_ branch on 20.08.2018).

Did anyone experience something similar once before? Any hint would be useful.

Thanks!

PatrickNa, 2018-08-28:
### How to avoid "message too big"  errors:

I have found a way to avoid this error and therefore the disconnection of the client. However it is **rather a hack than a solution**.

The default value of "max_message_size" (a parameter of the rosbridge_websocket launcher) is _None_, but just this value needs to be changed in order to advertise, publish or to subscribe to the server. After changing it to for example 1000, the publication reached the ROS machine.

Although the client is not disconnected any longer, I **still experience issues**:The server recognizes subscriptions to topics, but the published messages there never reach the client. I might open a separate issue for that and elaborate it further once I have better insights.

Maybe somebody else has an explanation for the default-value solution of mine and an idea of how to fix it in a proper way?

Schwimo, 2018-09-05:
I had this issue too and resolved it with: #353 
Downgrade tornado to version 4.5.2 with 
`sudo pip install tornado==4.5.2 --ignore-installed`

PatrickNa, 2018-10-01:
I will give it a shot. Thanks for the feedback. Appreciate it!

mvollrath, 2019-04-04:
Did this fix your problem?  Note there is a new TROUBLESHOOTING.md doc with some information about fixing your Tornado version.

PatrickNa, 2019-04-04:
Before I could change the Tornado version, the project switched back to Kinetic. Therefore I did no play around with it anymore.

Gil401, 2019-04-29:
Hi,
Did Anybody found a solution for this issue? encountered it too...

@PatrickNa  @mvollrath

PatrickNa, 2019-04-29:
Have you tried the downgrade of Tornado?

Gil401, 2019-04-29:
@PatrickNa now I did.
works. Thank you!!

However, is that the right way to go?

mvollrath, 2019-04-29:
Yes, using the distribution's packaged Tornado is the easiest way.  Adding compatibility for future Tornado versions is certainly possible, PRs would be welcome.

closed 2019-04-29 by mvollrath