RobotWebTools/rosbridge_suite #353: After tornado upgraded to version 5.0, websocket disconnects immediately after connection

zczcmlml, 2018-08-20:


florian-leblanc, 2018-08-29:
Thanks for your help man i didn't know why it doesn't working. This is now working with the version tornado==4.5.3

mojovski, 2018-08-29:
same here. Thanks !
You cant image how important this finding is. It is related many weird issues, people report with bson and apt-get ...

florian-leblanc, 2018-08-29:
Btw i use the version 4.5.3 but maybe more recent version are working before the 5.0.

kinergy, 2019-01-15:
> Yea i did something like `pip uninstall tornado` and then `pip install tornado==4.5.3`.

This fixed it for me on Ubuntu 16.04 with ROS Kinetic. Interestingly, Ubuntu 18.04 with ROS Melodic did not require this fix. Not sure why. My specific test case is running the RBSManager example app on iOS to control Turtlesim.

mvollrath, 2019-04-04:
Thank you for the troubleshooting, I'm adding information on Tornado version issues in #397

closed 2019-04-04 by mvollrath