RobotWebTools/rosbridge_suite #354: What \*is\* a ***correct*** timestamp for a message (particularly a STAMPED message)

eric-schleicher, 2018-08-20:
# TL;DR
**Defect:** 
RosBridge nukes valid header stamp values on received messages and replaces it with a value that is auto-generated (at receipt time).  While this behavior \*may\* be desirable in some general scenarios, it's absolutely the **wrong** thing to do on Stamped message types where the headers/stamp are exposed specifically because time is an integral component of the message data.  

**Impact:**
Stamped datatypes cannot be submitted vis rosbridge without data corruption.

I would like to fix it.  need advice on how 

---
I had asked a question some time ago in an [issue](https://github.com/RobotWebTools/roslibjs/issues/255) on a related project (roslibjs) about asserting timestamps in headers when using rosbridge.  The use case centered around the ability to provide very time-accurate sensor data to a ROS instance via rosbridge where the *real* canonical time of the sensor data was known prior to transmission or receipt (by rosbridge).  

At the time it was mentioned that if a valid timestamp was presented in an incoming message that rosbridge would (by default) copy the values over as it was internally creating the instance of the ros time value.  That issues references the following [code as the responsible section](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py#L222-L243) for this functionality.
```
def _to_time_inst(msg, rostype, inst=None):
    # Create an instance if we haven't been provided with one
    if rostype == "time" and msg == "now":
        return rospy.get_rostime()

    if inst is None:
        if rostype == "time":
            inst = rospy.rostime.Time()
        elif rostype == "duration":
            inst = rospy.rostime.Duration()
        else:
            return None

    # Copy across the fields
    for field in ["secs", "nsecs"]:
        try:
            if field in msg:
                setattr(inst, field, msg[field])
        except TypeError:
            continue

    return inst
```

In further trouble shooting positioning data within my system, I believe that I've discovered and validated this is in fact *NOT* the actual system behavior.  When submitting Stamped Message types via ros bridge (like _PoseStamped_) the values provided are replace with the rostime \*when\* the message is received.

In a StampedPose, the stamp value of the header really is intended to be the canonical representation of the time for which the message (independent of transience through the system).  

Proposed fix:

Regarding the [following code](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosbridge_library/src/rosbridge_library/internal/message_conversion.py#L272-L282) in `message_conversions.py` :
```
def _to_object_inst(msg, rostype, roottype, inst, stack):
    # Typecheck the msg
    if type(msg) is not dict:
        raise FieldTypeMismatchException(roottype, stack, rostype, type(msg))

    # Substitute the correct time if we're an std_msgs/Header
    try:
        if rostype in ros_header_types:
            inst.stamp = rospy.get_rostime()
    except rospy.exceptions.ROSInitException as e:
        rospy.logdebug("Not substituting the correct header time: %s" % e)

    inst_fields = dict(zip(inst.__slots__, inst._slot_types))
```
#### propose the following changes:
- change the comment to indicate that substituting receive time data is not 'correcting' data
- change the behavior to test if there is (valid) data in the header and to leave it.

## Questions
Is this something that i should just change and submit a merge request?  or is there a need to set this up to require some additional paramters for the rosbridge to minimize the chance of breaking other system which might accidentally rely on this error.

closed 2018-10-29 by eric-schleicher

arszero, 2018-12-10:
@eric-schleicher Hi, have you solved this issue? I am also trying to keep the original timestamp without replacing it with the receipt time. I tried to remove the substitution code in message_conversions.py but then all the timestamps I received became 0s. Is there any other part of the code that we need to modify?

eric-schleicher, 2018-12-20:
If you have control over the process creating the messages and the type of message, you can change to an 'Unstamped' Message type...   e.g   Pose Vs PoseStamped.  It appears that the handling for these are correct if they are populated.  that was the short term workaround.  I haven't looped back around to fix.

Theuns-Botha, 2021-02-17:
Hi there, I see this issue was closed without being resolved what is the reason for that? Or is there some other existing solution?

bochen87, 2022-04-29:
hello.
I noticed that somehow this behavior does not occur anymore. I put some debugging prints and noticed the time values get overridden later on again:
```
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:188]: custom_msgs/WebDriveCommand
message_conversion.py:190]: []
message_conversion.py:191]: {'header': {'frame_id': 'site_origin', 'seq': 0, 'stamp': {'secs': 1651218656, 'nsecs': 424999952}}, 'speed': 0, 'steering_angle': 0}
message_conversion.py:214]: End
message_conversion.py:188]: std_msgs/Header
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['header']
message_conversion.py:191]: {'frame_id': 'site_origin', 'seq': 0, 'stamp': {'secs': 1651218656, 'nsecs': 424999952}}
message_conversion.py:214]: End
message_conversion.py:290]: cur_time: 54544 825000000
message_conversion.py:294]: inst: 54544 825000000
message_conversion.py:188]: string
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['header', 'frame_id']
message_conversion.py:191]: site_origin
message_conversion.py:188]: uint32
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['header', 'seq']
message_conversion.py:191]: 0
message_conversion.py:188]: time
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['header', 'stamp']
message_conversion.py:191]: {'secs': 1651218656, 'nsecs': 424999952}
message_conversion.py:188]: float32
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['speed']
message_conversion.py:191]: 0
message_conversion.py:188]: float32
message_conversion.py:189]: custom_msgs/WebDriveCommand
message_conversion.py:190]: ['steering_angle']
message_conversion.py:191]: 0

```