RobotWebTools/rosbridge_suite #357: detect Disconnect of all Clients

AdronTech, 2018-09-21:
Is it possible to detect a disconnect on the Robot (Server)?

I would like to stop all the movement if the Gui disconnects due to signal loss.

kevinNT2018, 2019-03-25:
Hi @AdronTech, have you got any solutions to such a problem? Thanks for sharing.

mvollrath, 2019-03-25:
The rosbridge server should unsubscribe topics when there are no clients subscribed, depending on which client your underlying nodes are using you can detect when there are no subscribers.

For rospy there's [SubscriberListener](http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers#rospy.Publisher_initialization)

For roscpp there's [SubscriberCallbacks](http://docs.ros.org/melodic/api/roscpp/html/classros_1_1Publisher.html#ab5bb5f6fe398b32d0dae390e996968b6)

AdronTech, 2019-03-25:
Apparently the rosbridge_server websocket publishes the client_count, you could use that to detect a change in the client_count.
I have not tried it jet, but i think it should work.

viktorku, 2019-03-26:
You can solve it back-end/server wise from your logic. If you use roscpp, you can check number of subscribers on a publisher level [`publisher.getNumSubscribers()`](http://docs.ros.org/melodic/api/roscpp/html/classros_1_1Publisher.html#af5574532bf8a9b35efa3696a035d31fc) or [`publisher.get_num_connections()`](http://docs.ros.org/kinetic/api/rospy/html/rospy.topics.Publisher-class.html) for rospy.

If you need to trigger some extra logic from the client when it disconnects from rosbridge/websocket altogether, then use the `/connected_clients` published by rosbridge.

I'll close the issue as it's not really an issue but something that should belong to StackOverflow.

closed 2019-03-26 by viktorku