RobotWebTools/rosbridge_suite #358: Connection to websocket server closed

MathiasStrohkirch, 2018-09-21:
Hello, I am having an issue where the websocket server closes when I try to publish a message. I am using the code found here : http://iguanatronics.com/simple-tutorial-on-rosbridge-and-roslibjs/. NOTICE : "cnd" in lines 12 and 13 must be changed to "static". I am running roscore, turtlesim_node, and launching rosbridge. When I open the file in my web browser, it connects to the websocket server just fine, but disconnects when I click the "Publish Message" button.

ROS version 1.14.3 melodic
Ubuntu version 18.04.1 LTS bionic

rvipin17, 2018-10-23:
Hi, I'm facing the same issue, do you have any solution so far?

hoquangnam45, 2018-11-10:
Same, don't know why it happened

hoquangnam45, 2018-11-10:
What tornado version are you using? Turn out that i have 5.0 version which cause it to disconnect immediately after connect, try install tornado 4.5.3 and see how it goes

juliangaal, 2018-11-23:
Fixes it, thank you!

rabbishuki, 2019-03-28:
@hoquangnam45 That fixed it for me too,
Apparently tornado was already installed on a 5+ version.

Here are the steps to fix it if you have the same problem.

1. Check which version you have if you are running Ubuntu.
```
$ python
>>> install tornado
>>> tornado.version
<Your Version will be printed here>
```

2. If it is higher than 4.5.3
```
$ pip uninstall torando
$ sudo apt install ros-kinetic-rosbridge-suite
```

mvollrath, 2019-04-04:
Thank you for the troubleshooting, I've added a doc about this problem in #397

closed 2019-04-04 by mvollrath

CocoaG33k, 2019-09-17:
Ran into the same issue and found the core of the issue (at least for me) was related to this error "WebSocket connection closed: Code=1009, Reason=message too big", which is due to a misconfiguration in rosbridge_websocket.launch. See https://stackoverflow.com/questions/53965987/why-do-i-receive-message-too-big-on-publish-error-1009

juliangaal, 2019-09-18:
You mean [this fix](https://stackoverflow.com/a/54656254) specifically?