RobotWebTools/rosbridge_suite #363: Can Have Some Samples For Publish/Subscribe To Video/IMU

ybchoo, 2018-11-02:
I have been hunting hi + low to locate sample like those residing on 
https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes

I am trying to using RosSharp to subscribe to Video and IMU to remote machine.
However, I can find suitable example to subscribe to topic + message on ROS.

Can some kind soul out there please kindly help to provide some workable solution??

ybchoo, 2018-11-02:
Calling
      string x = this.instance.rosSocket.Subscribe<sensor_msgs.Image>
         ("/camera/color/image_raw", "sensor_msgs/Image", 
            subscribeImageHandler);

Function
   private  void subscribeImageHandler(sensor_msgs.Image image)
   {
      Debug.Log("Within Subscribe Image   [["+image.data +"]]");
   }
This keeps giving me 
Argument 2: cannot convert from 'string' to 'RosSharp.RosBridgeClient.SubscriptionHandler<RosSharp.RosBridgeClient.Messages.Sensor.Image>'
Can someone please point me to area where I can seek help? Thanks.

sevenbitbyte, 2018-11-02:
What approach are you using to access rosbridge from unity? I see several unity client packages for ROS and rosbridge maintained outside of RWT. Know exactly what clients you are using would be helpful in understanding the source of this issue. Do you have other topic publishers or subscribers that are working?

ybchoo, 2018-11-04:
Hi SevenBitByte
We only use ROS# by Martin_Bischoff. No other ROS packages are installed.
We follow strictly to https://github.com/siemens/ros-sharp/wiki/Info_CodeExample
This example work on Dot Net stand alone but when we use it on Unity, it does not 
activate the method **subscribeImageHandler**
Any pointer to solve this is greatly appreciated.

sevenbitbyte, 2018-11-05:
This sounds very much like an issue between Unity and ROS# rather than rosbridge_suite. Perhaps @MartinBischoff can help triage?

rosbridge_suite is responsible for the server side which sounds to be working as expected if you're code worked as standalone .NET

ybchoo, 2018-11-06:
Thanks. Case close.

MartinBischoff, 2018-11-06:
Thanks @sevenbitbyte for forwarding this issue. I already helped @ybchoo via personal mails.
For ROS# related issues, there is a dedicated board [here](https://github.com/siemens/ros-sharp/issues).

closed 2019-04-04 by mvollrath