RobotWebTools/rosbridge_suite #366: Silent python3 exception

clynamen, 2018-11-06:
The messages received over websocket are ignored silently due this exception:

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/tornado/websocket.py", line 547, in _run_callback
    result = callback(*args, **kwargs)
  File "/home/clynamen/ros/main_ws/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py", line 114, in on_message
    self.protocol.incoming(message)
  File "/home/clynamen/ros/main_ws/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py", line 120, in incoming
    self.buffer = self.buffer + message_string
TypeError: can only concatenate str (not "bytes") to str
```

The exception disappears silently within the _receive* coroutine of tornado

temporary fix, I will open a pull request once I dig the problem better:

https://github.com/RobotWebTools/rosbridge_suite/compare/develop...clynamen:python3_str_byte

**packages**:

- 
- python 3.7
- python-tornado 5.1.1
- ros melodic

mvollrath, 2019-03-05:
Is this still an issue?  Does the new websocket exception handling at least print the exception?

mvollrath, 2019-04-04:
Tornado 5.1.1 is not supported, see #371 .  Closing this for now, if there is still an issue let us know.

closed 2019-04-04 by mvollrath