RobotWebTools/rosbridge_suite #367: CBOR compression summary

mvollrath, 2018-11-08:
### PRs

- [ ] https://github.com/RobotWebTools/rosbridge_suite/pull/364
- [ ] https://github.com/RobotWebTools/rosbridge_suite/pull/365
- [ ] https://github.com/RobotWebTools/roslibjs/pull/303
- [ ] https://github.com/RobotWebTools/ros3djs/pull/239

### Background

Previous to these changes, there are two options for encoding data between rosbridge and clients: JSON and JSON inside a base64-encoded PNG image.

Both of these options are text-based.  Human-readable messages are great, but when the message data is thousands of integer values, the value of human readability diminishes and decoding speed becomes a bigger concern.  When visualizing messages containing large blobs of binary data such as PointCloud2 or OccupancyGrid, the existing methods don't scale.

With the goal of increasing both the **capacity** and **speed** of rosbridge/roslibjs encoding and decoding, I started looking for a binary encoding format that could a) comply to the rosbridge protocol and b) quickly decode numeric arrays as views of a JavaScript `ArrayBuffer`.

[CBOR](https://tools.ietf.org/html/rfc7049) is a perfect fit.  It implements a superset of JSON, adds byte arrays, and is extensible with tags.  With the addition of the [draft typed array tags](https://tools.ietf.org/html/draft-ietf-cbor-array-tags-00), it meets all of the requirements.  CBOR libraries are available for most languages and I found fast implementations for Python and JavaScript.

### Benchmarks

Before implementing CBOR compression I wanted to make sure it was actually faster than JSON or PNG compression.  I set up a Node.js application to simulate encoding and decoding some ubiquitous ROS message types containing binary data: sensor_msgs/Image and sensor_msgs/PointCloud2.

640x480 RBG Image:
```
CBOR encode: 1.523ms
JSON encode: 269.129ms
CBOR size: 921711 bytes
JSON size: 11474451 bytes
CBOR decode: 0.672ms
JSON decode: 95.300ms
```

PointCloud2 with 1928 points @ 32 point_step:
```
CBOR encode: 0.201ms
JSON encode: 12.775ms
CBOR size: 62001 bytes
JSON size: 703239 bytes
CBOR decode: 0.174ms
JSON decode: 6.152ms
```

After my proof-of-concept integration with rosbridge and roslibjs, I ran some more tests on a real application to verify that it would increase frame rate in a "real world" setting where message decoding competes with rendering for JavaScript time.  I compared the average decoding time for the first 100 messages of a 5,000 point PointCloud2:

```
JSON: 0.83ms
CBOR: 0.1ms
PNG: 45.37ms
```

Note that PNG didn't actually spend 45ms on the main thread because it decodes asynchronously, this was the "wall time" from message receipt to the handler being run.

A 80,000 point PointCloud2 demonstrates how much better binary encoding can scale:

```
JSON: 10.99ms
CBOR: 0.11ms
PNG: 264.48ms
```

Finally, I ran a "stress test" to see how a ROS3DJS PointCloud2 compares to rviz.  I made a [small patch to ROS3DJS to enable CBOR encoding with fast buffer-flipping](https://github.com/RobotWebTools/ros3djs/pull/239).  I was able to view a 250,000 point cloud with no frames dropped.  The same application with JSON or PNG decoding dropped at least one frame per message at this scale.  When I increased the cloud density to 500,000 points there was a visible frame drop with CBOR.  I noticed a similar frame drop in rviz on the same cloud.

To summarize, when CBOR compression is used, ROS3DJS has the same capacity for rendering point clouds as rviz.

### Implementation

CBOR works best when transmitted as a binary web socket message, so I took some liberty with the protocol.  Instead of sending a JSON message with a base64 string of CBOR data, the server sends an unwrapped binary protocol message.

On the client side, this message is received as an `ArrayBuffer` and the CBOR decoder is run on it directly to produce an object, just like the JSON decoder.  From there, everything works the same.

I produced some code to support typed array tags:
* On the server side, [cbor_conversion.py](https://github.com/EndPointCorp/rosbridge_suite/blob/cbor_encoding/rosbridge_library/src/rosbridge_library/internal/cbor_conversion.py) extracts CBOR-friendly values from a ROS message including tagged typed arrays.
* On the client side, [cborTypedArrayTags.js](https://github.com/EndPointCorp/roslibjs/blob/cbor/src/util/cborTypedArrayTags.js) defines a "tagger" for the decoder to efficiently unpack tagged arrays.

### Backwards Compatibility

Since CBOR is completely opt-in (the server will never send a CBOR binary message unless the client requests it), clients without CBOR decoding support are safe from newer server versions.

If the client requests CBOR but it is not available on the server side, the server will send JSON messages instead.

### When to use CBOR

CBOR is not ideal for all messages, but it's safe to say that it's the best choice for any message containing a large amount of binary data when using roslibjs in a browser.

JSON is still faster and better for small messages such as transforms, or messages containing mostly strings.

PNG is still *potentially* the smallest wire size where transport compression is not available, depending on the data.  When the data is mostly redundant, e.g. an array full of zeros, PNG messages *can* be smaller and decode faster than CBOR, because the deflate algorithm is efficient in this case, but when the data is more random, PNG will often have a *larger* wire size and longer decoding time than regular JSON compression.  This makes it a poor choice for PointCloud2 and situationally useful for Image or OccupancyGrid.

When wire size is a concern (not connecting to localhost), WebSocket per-message deflate on top of CBOR is the most efficient transport of binary messages from rosbridge to a browser.  When using per-message deflate, the decompression is handled natively by the browser and does not block the JavaScript thread.

mvollrath, 2018-12-05:
All merged :+1:

closed 2018-12-05 by mvollrath

eisenWu, 2019-10-23:
Hi @mvollrath ,

this is the exactly problem what i met, i use rosbridge send the point cloud to Unity and it is JSON, which is too slow and lagging, i am new to ROS , as i understood, if i want to use CBOR, i have to write a subscriber in ROS side to recive the point cloud data as JSON and a publisher to republish the data with CBOR, and wirte another subscriber in Unity side to recive the CBOR data. is it right to use the CBOR like this?

thanks in advance

Wei

mvollrath, 2019-10-23:
Hi Wei,

To get PointCloud2 data from ROS to Unity over rosbridge using CBOR you shouldn't need to write any ROS subscribers, rosbridge will do the CBOR encoding for you.

From Unity you will connect to the rosbridge socket and send a ["subscribe"](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe) message with the "compression" field set to "cbor".

Whenever a point cloud message comes into Unity over the socket, it will be a CBOR-encoded binary ["publish"](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#343-publish--publish-) message.

eisenWu, 2019-10-24:
Hi @mvollrath  thanks for the answer.
i use the [ROSbridgeLib](https://github.com/MathiasCiarlo/ROSBridgeLib) in Unity to connect with ROS via websocket, but i don't know how to send a "subscribe" message with the "compression" field set to "cbor".
what now i understand is, i add the additional "compression: "cbor"" in my subscriber in Unity, and the publicher in ROS will send the message in binary, namely i will recive the point cloud data in binary in Unity.

mvollrath, 2019-10-24:
ROSBridgeLib does not support CBOR decoding.  You have a few options:

* Get ROSBridgeLib to support CBOR decoding, either by creating an issue on their tracker and getting help or adding it yourself.

* You might also try [ros-sharp](https://github.com/siemens/ros-sharp), which also doesn't yet support CBOR decoding but has an issue tracker.

* Instead of using ROSBridgeLib, use a websocket library and send/receive the protocol messages yourself, decoding either JSON or CBOR as it comes in.

Sorry about this, CBOR encoding on the server side is a new feature and the client libraries haven't caught up yet.

eisenWu, 2019-11-18:
Hi @mvollrath ,
i am now getting start to deal with the CBOR, and trying to add a compression in ROSBridgeLib.
Could you tell me if i should write some code in ROS side in  oder to compress the message? i have tried to send message 
`{"op": "subscribe", "topic": "/zed/zed_node/point_cloud/cloud_registered", "type": "sensor_msgs/PointCloud2","compression": "cbor"}` 
but it seems nothing changed, maybe i did something wrong. 
Thanks in advance

mvollrath, 2019-11-18:
@eisenWu that looks correct, make sure you're running at least rosbridge_suite version 0.11.3.  You should get binary messages.  Compare to "compression": "json" which should yield text messages.

eisenWu, 2019-11-19:
Hi @mvollrath ,
today i did the test, and i found that when i want to get the byte[] Message from the function `OnMessage` in websocket-sharp, it contains a `RawData`. when i debug.log the `RawData`  it is a byte array. 
Now i am confused, even when i deleted the `"compression": "cbor"` it was aways byte array Message, and i have no idea if the CBOR works or not, how can i detect is the Message compressed?

mvollrath, 2019-11-19:
That `RawData` is always going to be a byte string, when you expect JSON try decoding it with UTF-8 into a string.

e.g. `string text = Encoding.UTF8.GetString(msg.RawData)`

eisenWu, 2019-11-20:
Hi @mvollrath ,
Thanks for your answer :)
i have send follow different msgs to ROS:
`{"op": "subscribe", "topic": "/zed/zed_node/point_cloud/cloud_registered", "type": "sensor_msgs/PointCloud2","compression": "cbor"}`
`{"op": "subscribe", "topic": "/zed/zed_node/point_cloud/cloud_registered", "type": "sensor_msgs/PointCloud2","compression": "none"}`
when i check the `RawData.length` , they have the same length of the byte Message, dose it mean i did not get CBOR from ROBBridge ?

eisenWu, 2019-11-20:
Hi @mvollrath ,
thanks for your answer :)
I think i have made a mistake, now once when i send 
`{"op": "subscribe", "topic": "/zed/zed_node/point_cloud/cloud_registered", "type": "sensor_msgs/PointCloud2","compression": "cbor"}`
to ROS, the window shows the infomation: Client disconnected. i use the ROS version: melodic, but have no idea how to see which version of rosbridge_suite i am using.
I do not have the decode of CBOR in my code yet, would that be the reson why the Client just disconnected?

mvollrath, 2019-11-21:
That is possible, maybe something to do with how C# `Encoding` handles illegal UTF-8 codes which may be present in a large blob of point cloud data.  Hopefully you can get some helpful log output from the client.  Unity prints traces in the player log if not in the editor log.

If your version of rosbridge_suite didn't support CBOR it would just send JSON.

zdzhaoyong, 2021-07-07:
> ROSBridgeLib does not support CBOR decoding. You have a few options:
>
> * Get ROSBridgeLib to support CBOR decoding, either by creating an issue on their tracker and getting help or adding it [...]

I was confusing why the rosbridge server only send CBOR but not receive it. 
The CBOR decoding should not a difficult job.

zdzhaoyong, 2021-07-07:
I just wrote a C++11 based rosbridge client where CBOR (de)serializer supported.
See. https://github.com/zdzhaoyong/svar_rosbridge
By using [Svar](https://github.com/zdzhaoyong/Svar), it can be used by both C++ and Python.
Hope the rosbridge_server also supports CBOR decoding soon!

wyq730, 2022-02-08:
@mvollrath 

Thanks for this doc. That's a great introduction!

I'm wondering, however, how is CBOR compared with some common bytes-compatible serialization schemes like [bson](https://bsonspec.org/) or [msgpack](https://msgpack.org/index.html)? I did a simple benchmark and it seems that CBOR didn't do better than them, so I would like to know if there is additional rationale behind which I didn't take into consideration. Thanks!

mvollrath, 2022-02-09:
The thing I was looking for was performance at homogenous typed arrays, e.g. large arrays of int16 being transformed into JavaScript typed arrays.  There's a big difference between a CBOR array and a tagged typed array; instead of declaring the type of each array element you just put a tag in front and pack the bytes as densely as possible.  Decoding the bytes is very efficient in most languages incl. JavaScript (typed arrays from byte strings is superfast) and Python (import struct and unpack is superfast).  These kinds of homogenous arrays are used in ROS messages like point clouds and occupancy grids and such, and these are the kinds of messages CBOR encoding was intended for.  JSON is still faster in typical browsers for small messages or messages with many keys because it has native implementation.

Otherwise:
* BSON was never designed to be a wire protocol and all the libs I tried were some combination of slow, inefficient, or cumbersome.
* Msgpack is what CBOR is based on, but didn't seem to have the same kind of tagging scheme to efficiently deal with packed arrays.
* CBOR is an IETF standard, unlike the other two which are proprietary protocols.

Hytac, 2023-05-08:
Hi, i'm having the following error when using cbor for OccupancyGrid

`[rosbridge_websocket-1] [INFO] [1683543613.742826618] [rosbridge_websocket]: [Client 1df4abbf-5909-46b4-9923-855a7d997bec] Subscribed to /map_manager/occupancy_grid
[rosbridge_websocket-1] [ERROR] [1683543614.276534924] [rosbridge_websocket]: Exception calling subscribe callback: 'OccupancyGrid' object has no attribute '_slot_types'
`
Any idea?

jesusramondovale, 2023-10-23:
> Hi, i'm having the following error when using cbor for OccupancyGrid
>
> `[rosbridge_websocket-1] [INFO] [1683543613.742826618] [rosbridge_websocket]: [Client 1df4abbf-5909-46b4-9923-855a7d997bec] [...]

I'm having exactly the same issue