RobotWebTools/rosbridge_suite #370: BSON mode slow with arrays

jteuber, 2018-11-14:
I'm working on a simulation project that needs to transfer large amounts of data comprised of different message types to ROS via the ROSbridge. Because of that I'm using the BSON-only-mode as described in #257 . 
After doing some tests with a custom message type and sensor-msgs/LaserScan (with 1080 range-values) I found that the ROSBridge can't keep up with the flood of messages (I'm sending the messages with 40-100Hz). My best guess would be that the packing and unpacking of the arrays into BSON is the problem as every value is prefixed with the index. Is this really necessary?
Would it be possible to send arrays as binary blobs as in sensor_msgs/Image instead?

I would be willing to implement that myself, but I would need a pointer to the spot in the ROSBridge code where this would need to be changed.

mvollrath, 2018-12-05:
You might try the new "cbor" compression on develop, see https://github.com/RobotWebTools/rosbridge_suite/issues/367 .  When I evaluated BSON for binary ROS messages I found that the JS decoder was very slow and the message size was larger than JSON.

mvollrath, 2019-01-15:
For BSON the JS decoder is used by roslib to decode the binary message on the browser side.

[Here is some old benchmark stuff](https://github.com/mvollrath/cbor-js), I took out the BSON testing and never committed the version that tests encode/decode multiple times (only testing the first call is a flawed benchmark) but it's a start.

mvollrath, 2019-02-06:
> @mvollrath I just noticed that your cbor implementation in rosbridge was done after rosbridge was ported to ROS2, is that correct?

No, it is in kinetic and I've only tested it with ROS1.  If you have rosbridge_suite at least 0.10.x it has CBOR encoding.

mvollrath, 2019-04-04:
Closing this for now, if you have performance issues with BSON you should use CBOR.

closed 2019-04-04 by mvollrath