RobotWebTools/rosbridge_suite #371: Tornado ~5.1.1 breaks websocket connection

juliangaal, 2018-11-23:
Tornado (I was running 5.1.1) seems to immediately drop the connection to the websocket, as soon as you try to send the first message. 

Error on first message: `invalid state`

Error on 2-n: `Websocket connection closed`

Installing tornado 4.5.3 fixes the issue.

Originally posted by @hoquangnam45 in https://github.com/RobotWebTools/rosbridge_suite/issues/358#issuecomment-437566434_

fudanglp, 2019-01-05:
Same issue here.
I have seen memory leak in rosbridge_websocket and wanna debug it.
I doubt its tornado's cause so I upgraded tornado and same thing happened

mvollrath, 2019-01-19:
You might try the solution in #378

juliangaal, 2019-01-19:
Doesn't work. Had to downgrade tornado

mvollrath, 2019-04-04:
You need to use the version of Tornado installed by rosdep, tracked by rosdistro.  Any other version of Tornado is not supported.  If you need support for other versions of Tornado, feel free to open a PR.

closed 2019-04-04 by mvollrath