RobotWebTools/rosbridge_suite #372: python-cbor package not available in trusty/indigo

mvollrath, 2018-12-11:
The package was introduced in xenial, so if any further indigo releases from develop are planned, I can add some logic to make the server not break at import time.

alecive, 2018-12-12:
Have you tried using the `indigo` branch? It doesn't seem to have this issue.

mvollrath, 2018-12-12:
Indigo build would likely break upon finding the unresolved rosdep in package.xml, the usual hack is inlining the dependency source in this repo.  The CBOR lib is lightweight and Apache licensed, if anybody needs CBOR in indigo, that's what I would do.

jihoonl, 2018-12-14:
Um actually I stop releasing indigo since version 0.7.16 due to the dependency management issue. So I wouldn't bother to keep the backward compatibility for indigo in develop and master branch.

closed 2018-12-14 by jihoonl