RobotWebTools/rosbridge_suite #378: Websocket closed before receiving a handshake respnse

uahic, 2018-12-14:
Since building and sourcing the latest github version of the rosbridge_server, it seems that connecting from roslib.js to rosbridge_server does not work on my system (using the old rosbridge_server there is no issue):

> WebSocket connection to 'ws://localhost:9090/' failed: Connection closed before receiving a handshake response

regarding netstat, port 9090 is open and served by a python process (rosbridge_server)
regarding a search on google this seems to happen if multiple processes want to serve this port (one possibility) but i only started it once

uahic, 2018-12-17:
I'm getting the same problem also when using the python implementation instead of roslib.js

mvollrath, 2019-01-04:
I had this problem and fixed it by clearing out a broken tornado module in my workspace.  `rm -rf build/ devel/` then rebuild.

mvollrath, 2019-04-04:
See TROUBLESHOOTING.md for solutions to common problems, does this help?

closed 2021-09-17 by github-actions[bot]